m-labs/artiq #541: Ver 1.3: cannot find -lcompiler_rt during gateware build

jasonamini, 2016-08-19:
Building gateware for artiq 1.3 with current misoc version returns "or1k-linux-ld: cannot find -lcompiler_rt" for target ksupport.elf. 

Command leading to error: python3.5 -m artiq.gateware.targets.kc705 -H qc1 --toolchain vivado

This seems to be the same issue that commit 929ded30b8dd5f0c9c9afd35c68c130ffc63be6c fixes on version 2. Testing the same changes on  1.3 (compiler_rt to compiler-rt in  artiq/runtime/Makfile) gets past this error.

jordens, 2016-08-19:
That commit does not touch anything related.

jordens, 2016-08-19:
Did you follow the manual when installing?
In particular, are the submodules checked out in misoc?

whitequark, 2016-08-19:
> Did you follow the manual when installing?
> In particular, are the submodules checked out in misoc?

Yes, I believe he did, that's why he got the error. misoc 0.1 should be used with artiq 1.3.

jordens, 2016-08-19:
Scratch that. `release-1`  needs to see that change as well. As a workaround use a misoc revision that does not have the change (https://github.com/m-labs/misoc/commit/d3c673599c50fe9a39e437f0c11d92fb0730631a).

whitequark, 2016-08-19:
@jordens Should we put misoc 0.1 into the main channel? I'm not sure why it isn't there already.

jordens, 2016-08-19:
@whitequark the "installing from source" instructions use misoc master. Either that needs to be changed (and retroactively noted for 1.X as well). Or `release-1` needs to see that change.

jordens, 2016-08-19:
@whitequark misoc is not needed to install artiq from the main channel. Just needed for building. But I'd be fine to *also* have misoc 0.1/0.2 in the main channel.

whitequark, 2016-08-19:
@jordens The install from source instructions for ARTIQ also use ARTIQ master.

jordens, 2016-08-19:
@whitequark Yes. But then their existence as part of the shipped release is at least questionable, right?

jasonamini, 2016-08-19:
I am now far behind the conversation and am probably hashing out things you have already figured out, but here are the answers in order:
1) Looks like I pasted in the wrong commit. This should be the "artiq/runtime/Makefile" change "runtime: follow rename of compiler_rt to compiler-rt in misoc" on 8/16/2016. Trying to paste in again: 283fc904ecf47a59bf512d2b6f4d32d410291fd1
2) I walked (mostly) step-by-step down the artiq-master version of the manual from your website. There are a couple of additional minor  fixes that I need to put in in regard to the manual, but I was able to get to the Vivado compile step in ver 2 (need to get the Vivado registration key transfered, so it stopped there)  before trying out ver 1 and running into this issue.
3) I will try out the earlier misoc version.
FYI: I am running a clean install of Ubuntu 16.04 and so far so good.

jasonamini, 2016-08-19:
Correction: I was able to get the pre-Rust version 2  (538d3e8a85db4f22939abacc07038dbb2b940102) to work. The last couple of commits for Rust broke my compile.

jasonamini, 2016-08-19:
The earlier version of misoc works OK with artiq 1.3.

sbourdeauducq, 2016-08-20:
> misoc 0.1 should be used with artiq 1.3.

No, MiSoC 0.2 (exactly at the tag, without the later commits) should be used with ARTIQ 1.x.

sbourdeauducq, 2016-08-20:
I have put MiSoC 0.2 and Migen 0.3 into the main channel. Use those for ARTIQ 1.x.

closed 2016-08-20 by sbourdeauducq