m-labs/artiq #542: Documentation updates for artiq-master

jasonamini, 2016-08-19:
In installing artiq v2 on a new system (Ubuntu 16.04), I found that I needed to deviate from the manual as follows:

1) The llvmlite version needed to be updated to 3.9 (from 3.8):
git checkout artiq-3.9

2)  Needed to install llvmlite to complete  "Preparing the core device FPGA board". The instructions, along with pygit2, are later in the manual. Move them earlier or did I mess up?

3) Ubuntu 16.04 differences:
-- pygit2 version is 0.24.1 (matching libgit2-dev version) and pip (pip3 is not installed by default) works fine with Python3.5: python3.5 `which pip` install --user pygit2==0.24.1
-- Needed to install cmake, cargo, and  libgit2-dev. Not absolutely sure if the last was needed, but I installed it to figure out the pygit2 version number. 
-- Joe Britton also recommended installing build-essential in case the system did not have the appropriate build tools selected during Ubuntu install.

whitequark, 2016-08-19:
> 1) The llvmlite version needed to be updated to 3.9 (from 3.8):
> git checkout artiq-3.9

Actually, this shouldn't be the case. I've rolled back all 3.9 changes. What prompted you to switch to 3.9?

> 2) Needed to install llvmlite to complete "Preparing the core device FPGA board". The instructions, along with pygit2, are later in the manual. Move them earlier or did I mess up?

I think there's some issue where things are imported too eagerly.

> -- Needed to install cmake, cargo, and libgit2-dev. Not absolutely sure if the last was needed, but I installed it to figure out the pygit2 version number.

Why did you need libgit2-dev? There's already libgit2 as well as pygit2 packaged in conda. Cargo is in conda too (although the manual needs to be updated for that, I guess).

jasonamini, 2016-08-19:
1) I cannot compile llvmlite with version 3.8. Tried again to make sure:
_targets.cpp: In function ‘void LLVMPY_AddTargetData(LLVMTargetDataRef, LLVMPassManagerRef)’:
targets.cpp:95:29: error: ‘LLVMAddTargetData’ was not declared in this scope
targets.cpp: In function ‘LLVMOpaqueTargetMachine* LLVMPY_CreateTargetMachine(LLVMTargetRef, const char*, const char*, const char*, int, const char*, const char*, int, int)’:
targets.cpp:224:14: error: ‘Default’ is not a member of ‘llvm::Reloc’_
Works fine with 3.9.

2) OK

3) I should have been more careful in my statement: I downloaded libgit2-dev to figure out the correct version number and I assumed that anything wrapping git2 would need the dev install. I don't know if it was actually needed.

whitequark, 2016-08-19:
> targets.cpp: In function ‘void LLVMPY_AddTargetData(LLVMTargetDataRef, LLVMPassManagerRef)’:
> targets.cpp:95:29: error: ‘LLVMAddTargetData’ was not declared in this scope
> targets.cpp: In function [...]

Right, the doc should be changed to say `git checkout artiq-3.8` now.

jasonamini, 2016-08-19:
Don't you mean _git checkout artiq-3.9_? Version 3.8 did not work with the version of llvm I downloaded and compiled yesterday.

whitequark, 2016-08-19:
The LLVM and clang instructions should be similarly changed to 3.8, yes.

whitequark, 2016-08-21:
I believe I've addressed all this in cea57e2.

closed 2016-08-21 by whitequark

jasonamini, 2016-08-22:
Small issue with the new LLVM-or1k and Rust instructions. Need to change directories to the project after clone before you can checkout the the artiq-3.8 versions. For example:
_git clone https://github.com/m-labs/llvm-or1k
**cd llvm-or1k**
git checkout artiq-3.8
cd tools
git clone https://github.com/m-labs/clang-or1k clang
**cd clang**
git checkout artiq-3.8
cd ../..
mkdir build
cd build_
Same for Rust.