m-labs/artiq #543: Warning if kernel_invariant not found

r-srinivas, 2016-08-26:
I have the following code here,

```python
from artiq.experiment import *

class Timing_Test(EnvExperiment):
    '''Timing Test'''
    kernel_invariants = {'y'}        

    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl0")
        self.x = 100*ns

    @kernel
    def run(self):
        for i in range(1):
            rtio_log("ttl0", "i", i)
            for j in range(20):
                self.ttl0.pulse(self.x)
                delay(self.x)
```

Which runs fine, but the kernel invariant was mistyped. However, there's no error or warning raised. Calling `self.x` in the kernel is still an expensive operation that costs a lot of slack. It could be useful to generate an error message or warning if a kernel_invariant does not match any variable, in case the user mistypes its name.

whitequark, 2016-08-27:
Unfortunately, this is not possible. The reason is that which code will be compiled into a kernel is only discovered at runtime. Consider the real case of `TTLClockGen`. It has `self.acc_width` marked as kernel invariant, however if only `set_mu` and not `set` is ever called on it, no code that uses `self.acc_width` will be compiled, or even discovered.

Eagerly typechecking all code marked `@kernel` would be prohibitively slow.

whitequark, 2016-08-27:
Something we *could* do is add a static analyzer that, when explicitly ran, suggests which variables could be marked as kernel invariants.

sbourdeauducq, 2016-08-27:
Impossible? What about iterating through kernel_invariants and checking that attributes exist, printing a warning if they don't (maybe with a bit of Levenshtein)?

whitequark, 2016-08-27:
@sbourdeauducq This means that variables accessible to kernels should be never conditionally set. If this is an acceptable restriction, sure, I can implement that.

sbourdeauducq, 2016-08-27:
With conditionally set variables, I think that having a spurious warning that you can suppress by giving the variable a dummy value (e.g. in ``build``) is totally fine.

sbourdeauducq, 2016-08-27:
And there would be no typechecking so that dummy value can always be ``None``.

jordens, 2016-08-27:
This is a band-aid and one can question whether it is worth applying. There are many ways a spelling error can go unnoticed here. You could misspell `kernel_invariants` or the spelling error could lead to the wrong variable being in `kernel_invariants`.
Wouldn't the real solution be some more "static" typing, along the lines what Enthought traits did or what `attrs` does, like `x = attr.ib(invariant=True)` in the class body?

whitequark, 2016-08-27:
> This is a band-aid and one can question whether it is worth applying.

My primary concern here is false positives, since an analysis with lots of false positives in practice just means that people ignore it (or turn it off). It is no a monumental amount of work and if it's good false positive wise then I see no reason not to implement it.

> Wouldn't the real solution be some more "static" typing, along the lines what Enthought traits did or what attrs does, like x = attr.ib(invariant=True) in the class body?

Unless I'm missing something, you can still misspell `x` here...

jordens, 2016-08-27:
I am fine with applying a band-aid if it's not much work and if there is a benefit.

The idea would be to enforce that all attribute use by kernels must be typed like this. Then you can't misspell it.

whitequark, 2016-08-27:
> The idea would be to enforce that all attribute use by kernels must be typed like this. Then you can't misspell it.

If we make annotations on attributes that are referenced in kernel code mandatory, then that comes with a large benefit: we can make user-defined objects parameterizable. E.g. a hypothetical TTLWrapper would admit both TTLIn and TTLInOut objects as a value of one of its attributes. It would not be a burden to implement at all, since all the plumbing is already there (and is used for e.g. `list`), but adding this without explicit annotations would just be an awful experience.

r-srinivas, 2016-08-29:
> Something we could do is add a static analyzer that, when explicitly ran, suggests which variables could be marked as kernel invariants.

I think that's probably useful for the debugging we'd like to do. Basically if we had some tool that could let us know what was kernel invariant and what isn't, we could improve the code to reduce the demands during the experiment.

commit feeb0895 (2016-09-14): compiler: warn about unused kernel_invariant entries.

Fixes #543.
commit ecb01d05 (2016-09-15): compiler: warn about unused kernel_invariant entries.

Fixes #543.
closed 2016-09-14 by whitequark