m-labs/artiq #545: Can't call GUI arguments in build

r-srinivas, 2016-08-31:
Using (2.0.dev+563.gc4559fc) on windows, the following code,

```python
from artiq.experiment import *

class GUI_Argument_Test(EnvExperiment):

    def build(self):
        self.setattr_argument('x', NumberValue(1))
        self.setattr_argument('y', NumberValue(2))
        self.z = self.x + self.y

    def run(self):
        print(self.z)
```

has the error,

```
TypeError: unsupported operand type(s) for +: 'NoneType' and 'NoneType'
ERROR:worker(examine,gui_argument_test.py):root:Terminating with exception (Type
Error: unsupported operand type(s) for +: 'NoneType' and 'NoneType')
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-08-31\lib\site-packages\artiq\master\worker
_impl.py", line 243, in main
    examine(ExamineDeviceMgr, ParentDatasetDB, obj["file"])
  File "C:\Anaconda3\envs\artiq-2016-08-31\lib\site-packages\artiq\master\worker
_impl.py", line 145, in examine
    exp_class((device_mgr, dataset_mgr, argument_mgr))
  File "C:\Anaconda3\envs\artiq-2016-08-31\lib\site-packages\artiq\language\envi
ronment.py", line 207, in __init__
    self.build(*args, **kwargs)
  File "c:\artiq-magtrap\repository\test\gui_argument_test.py", line 13, in buil
d
    self.z = self.x + self.y
TypeError: unsupported operand type(s) for +: 'NoneType' and 'NoneType'
```

This was working before in 1.x. I can move any operations into prepare, and then it works,

```python
from artiq.experiment import *

class GUI_Argument_Test(EnvExperiment):

    def build(self):
        self.setattr_argument('x', NumberValue(1))
        self.setattr_argument('y', NumberValue(2))

    def prepare(self):
        self.z = self.x + self.y

    def run(self):
        print(self.z)
```

But it's useful I think to be able to manipulate gui arguments in the build section if you're constructing dependent gui variables.

sbourdeauducq, 2016-09-01:
Yes, in 2.0 all arguments return ``None`` during scan/examine (to build the GUI).
What exactly are dependent GUI variables? Can't your code be written in another way?

r-srinivas, 2016-09-01:
I had some microwave code for example that defaulted the pi time based on what electrode was selected,

```python
        self.setattr_argument("t_pi", NumberValue(
                              self.get_dataset("mw.t_pi.3p2_2p2" + ".e" +
                                               str(self.mw.electrode)),
```

But I suppose that `self.mw.electrode` comes from the default value so I can work around these. There were other instances where I determined the actual frequency to send to the dds based on the gui arguments,

```python
        self.d0 = self.get_dataset("raman.light_shift")
        self.setattr_argument("axial_frequency",
                              NumberValue(self.get_dataset("raman.stretch.f"),
                                          scale=1e6, unit="MHz", ndecimals=3),
                                          "Scan")
        self.f0 = (self.get_dataset("mw.f.3p3_2p2") +
                   self.get_dataset("mw.d.3p3_2p2") +
                   self.d0 + self.axial_frequency)/4
```

But I suppose that could be moved into prepare. That's just how we've been doing it till now and it seemed to work. We can rewrite the code, but could this maybe be added to the release notes?

dleibrandt, 2016-09-01:
I've also been doing math on GUI arguments in build.  In addition to scenarios similar to @r-srinivas 's above, another scenario where I find it useful is similar to the `SubComponent1` class in `arguments_demo.py`.  I often want to do some math on the GUI arguments of something like `SubComponent1`.  Putting it in a method `SubComponent1.prepare()` is annoying because I would have to manually call this from the parent.  Putting it in `SubComponent1.do()` is no good if the math takes a long time and this is a kernel.  Thus, I have been putting it in `SubComponent1.build()`.

sbourdeauducq, 2016-09-02:
How about adding a way to tell if the experiment is being examined (not run) so you can skip the otherwise failing computations?
It would look like:
```python
 self.setattr_argument('x', NumberValue(1))
 self.setattr_argument('y', NumberValue(2))
 if not self.examining:
   self.z = self.x + self.y
```

r-srinivas, 2016-09-02:
When would that computation be executed then?

sbourdeauducq, 2016-09-02:
When the experiment is submitted and run.

r-srinivas, 2016-09-02:
This would mean you still couldn't do something like,

````python
self.setattr_argument('x', NumberValue(1))
self.setattr_argument('y', NumberValue(2))
self.setattr_argument('z', NumberValue(self.x+self.y))
```
Right? That's not a big deal, but would these means that such values won't be pre-computed then?

sbourdeauducq, 2016-09-02:
Write it like this:
```python
default_x = 1
default_y = 2
self.setattr_argument('x', NumberValue(default_x))
self.setattr_argument('y', NumberValue(default_y))
self.setattr_argument('z', NumberValue(default_x+default_y))
```

Using the returned argument values in ``build`` like you do is problematic: the order of the ``get_argument`` calls determines the order of the widgets, so you could only have the default value of a widget depend on the default values of the widgets above it, which is artificial. This is why 2.0 prevents this code pattern by giving you ``None`` for all arguments during scan/examine.

dleibrandt, 2016-09-02:
I like the `self.examining` idea.  To solve @r-srinivas 's first scenerio, one could do something like:
```
        if self.examining:
            default = 0
        else:
            default = self.get_dataset("mw.t_pi.3p2_2p2" + ".e" + str(self.mw.electrode)
        self.setattr_argument("t_pi", NumberValue(default ),
```

dleibrandt, 2016-09-02:
Although I suppose one could alternatively just check if `str(self.mw.electrode)` is None.

sbourdeauducq, 2016-09-02:
Datasets are accessible during scan/examine.
Proper testing for None is ``x is None``, no strings involved.

r-srinivas, 2016-09-02:
I should rewrite my code to take defaults as you suggested, but it would be useful to be able to calculate in `build` as @dleibrandt suggested. The `self.examining` idea is fine with me, as long as it doesn't affect slack during the experiment in anyway, which it seems like it shouldn't.

sbourdeauducq, 2016-09-02:
> it would be useful to be able to calculate in build as @dleibrandt suggested.

What exactly did he suggest that is not possible right now?

r-srinivas, 2016-09-02:
If something like this occurs in a subcomponent,

```python
        self.setattr_argument("axial_frequency",
                              NumberValue(self.get_dataset("raman.stretch.f"),
                                          scale=1e6, unit="MHz", ndecimals=3),
                                          "Scan")
        self.f0 = (self.get_dataset("mw.f.3p3_2p2") +
                   self.get_dataset("mw.d.3p3_2p2") +
                   self.d0 + self.axial_frequency)/4
```

It would raise an error. It would be nice to have it this way as opposed to calling the default value of `self.axial_frequency` if for some reason you want to change the frequency and scan at that value instead.

sbourdeauducq, 2016-09-04:
Well yes, but ``prepare`` is the intended location for such computations, to keep ``build`` light and fast.

sbourdeauducq, 2016-09-04:
How about changing the default ``Experiment.prepare`` so it automatically calls ``prepare`` on the experiment's children?

dleibrandt, 2016-09-06:
That sounds good to me.

r-srinivas, 2016-09-06:
I think so too. Basically as long as it's called automatically in the children, it doesn't matter whether it's in the `build` or `prepare` method to me.

dhslichter, 2016-09-06:
+1 from me as well for `prepare` calling any children's `prepare` as well, by default.

jordens, 2016-09-07:
The order in which they are called would be depth-first alphabetic (and not order of instantiation). And if you override `prepare()` you would need to take care of calling the parent's `prepare()` or do everything yourself.

sbourdeauducq, 2016-09-07:
We can do order of instantiation, the child just has to notify the parent.

commit 5fbcdace (2016-09-07): environment: call prepare for children by default, closes #545
closed 2016-09-07 by sbourdeauducq