m-labs/artiq #546: Repository scan slow

r-srinivas, 2016-08-31:
Using (2.0.dev+563.gc4559fc) on windows, a repository scan of a 100 of these files,

```python
from artiq.experiment import *

class Test0(EnvExperiment):

    def build(self):
        self.setattr_argument('x', NumberValue(1))

    def run(self):
        print(self.x)
```

which are identical except that the class name and file name were replaced by their index, takes about 55 s. 10 files takes about 5 s, 200 about 114s. Our actual experiment folder with about 84 files takes 63s or so. This can be a long time to wait before we can run any experiments if we have to restart artiq or make changes to the repository. Is there any way to speed this up or, is 0.5 s per file something we just have to live with? The numbers are about the same on the virtual machine.

I used the following script to generate the files,

```python
import os

f = open('c:\\repository_scan_test\\repository\\test0.py')
data = f.read()
f.close()
for i in range(1, 100):
    new_data = data.replace('Test0', 'Test'+str(i))
    f = open('c:\\repository_scan_test\\repository\\test{}.py'.format(i), 'w')
    f.write(new_data)
    f.close()
```

dhslichter, 2016-09-08:
@sbourdeauducq @jordens does it take a similar amount of time on your machines, or is this potentially another case of the overzealous antivirus slowing down creation of all of the worker processes that do the scanning of the experiments?

sbourdeauducq, 2016-09-08:
Rather slow here as well and there does not seem to be an obvious solution (e.g. recycling workers does not help much).

jordens, 2016-09-08:
Why does recycling workers not help much?

dhslichter, 2016-09-08:
This problem is only going to get worse as we build up more and more experimental code, and it is also something that first-time ARTIQ users will encounter rapidly and may find off-putting, so if there are straightforward ways to improve efficiency here it seems that would be very useful.

sbourdeauducq, 2016-09-09:
> Why does recycling workers not help much?

I don't know; this is simply the first thing I had tried and it did not seem to significantly improve things.

sbourdeauducq, 2016-09-09:
Actually it does, I probably was doing it wrong the first time I tried.

sbourdeauducq, 2016-09-09:
Should be faster by an order of magnitude.

closed 2016-09-09 by sbourdeauducq

r-srinivas, 2016-09-09:
Great, thanks. Could this be added to the 2.0 release?

commit 38768835 (2016-09-09): master: optimize repository scan, closes #546
commit ad697d43 (2016-09-09): master: optimize repository scan, closes #546