m-labs/artiq #549: GUI table argument (feature request)

dleibrandt, 2016-09-02:
It would be convenient to have a GUI table argument type.  I would use this for, among other things, a table of trap voltages where rows are the different electrodes and columns are different settings (used for different numbers of ions or other purposes).

(In the Ionizer, we have something like this based on QTableWidget.)

sbourdeauducq, 2016-09-03:
More details please.

dleibrandt, 2016-09-06:
By table, I mean a 2D array, that would be displayed in the GUI similar to a (fixed and finite size) spreadsheet.  Displaying row and column labels would be necessary.  Displaying units would be nice but not necessary.  I think that using the same type for all the elements would be fine, and all the use cases I can think of would want the type to be floats.  I believe that the QTableWidget object in qt does most or all of what I want in terms of GUI behavior.

Of course, it would be possible to use many NumberValue GUI arguments to accomplish the same thing, but displaying a table is much more space efficient, and it allows the user to easily compare the rows and columns since they are next to each other.

dleibrandt, 2016-09-06:
In the experiment code, this could be represented as either a 2D list or a 2D numpy array of floats, I don't really have a strong preference about this part.

sbourdeauducq, 2016-09-07:
Still more details are required...

How are the number of rows and columns defined? How are the row and column labels defined? Do all the numbers in the table have the same unit? Should we use ScientificSpinbox for editing the numbers? Should the editing widgets be active all the time (as if you were using many NumberValues) or popped up when double-clicking on a value in the table (same behavior as the applet table)?

dleibrandt, 2016-09-07:
I would require the number of rows and columns as well as the row and column labels as arguments to the `init` method of the `TableValue` argument processor.

It would be nice if the user had the option of specifying different units and scales for each element, or perhaps each row of the table, but if that is too much work than I'd be fine with making them all the same.  (In the trap voltages example we might actually be specifying gradients [V/m] and curvatures [V/m^2] of the electric field in the different rows.)

I like the ScientificSpinbox for editing the numbers.

I'd be inclined to have all the editing widgets active all the time.

Are there any more important details?

dhslichter, 2016-09-08:
I would vote that units/scale/step be selectable by row/column.  When providing `units`, `scale`, etc. for columns, rows, you can either provide a single value (which would be the same for all rows and all columns), or a list of values for each row and/or each column.  For an n x m table, the row units/scale/step/etc would need to be a single value or an n-element list, and the columns a single value or an m-element list.  

The default values, mins, maxes, etc would be either single element (for all in the table) or an n x m 2D list (or numpy array).  

one would have for example:
```python
# homogeneous table, arbitrary number of rows/cols
runits="V/m"
cunits="electrode #"

# inhomogeneous table, 2 rows, arbitrary number of columns
runits=["V/m", "V/m^2"]
cunits="electrode #"
````