m-labs/artiq #550: rtio_log documentation

dleibrandt, 2016-09-02:
(I'm running 2.0 on linux.)

I don't understand how the function `rtio_log` is supposed to work.  I'm trying to use it following the example in the manual (slightly modified):
```
from artiq.experiment import *

class RTIOlog(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl12")

    @kernel
    def run(self):
        for i in range(100):
            self.ttl12.pulse(1*ms)
            delay(1*ms)
            rtio_log("ttl12", "i", i)
            delay(1*ms)

```
I don't see the value of `i` anywhere in the vcd file.  Where does it go?  In any case, perhaps some more thorough documentation is in order.

sbourdeauducq, 2016-09-03:
Please post the problematic VCD file plus the raw dump (``artiq_coreanalyzer -d raw_dump -v vcd``).

dleibrandt, 2016-09-06:
The following was generated with `artiq_coreanalyzer -d rtio_test -w rtio_test.vcd`:
[rtio_test.zip](https://github.com/m-labs/artiq/files/457894/rtio_test.zip)

sbourdeauducq, 2016-09-07:
@whitequark the problem is the ``1d`` special character used for separating messages is not emitted or handled correctly by the logging system, and no log message ever gets terminated.

sbourdeauducq, 2016-09-07:
Is 1.x affected?

dleibrandt, 2016-09-07:
I see the same behavior in 1.2.

whitequark, 2016-09-13:
@sbourdeauducq I thought you were looking at the RTIO bug?

sbourdeauducq, 2016-09-13:
I asked you to characterize the gateware problem, if it really is one. You said that it dropped characters, but the gateware has no notion of character.

whitequark, 2016-09-13:
Yes, that seemed odd to me too. I'll look at it again shortly.

whitequark, 2016-09-14:
@sbourdeauducq This is not a gateware bug. It's actually a bug in `_extract_log_chars`. I will fix it shortly.

commit 494cfca4 (2016-09-14): coreanalyzer: fix rtio_log message extraction.

Fixes #550.
commit fb3585f5 (2016-09-14): coreanalyzer: fix rtio_log message extraction.

Fixes #550.
commit 3c8a7df9 (2016-09-14): coreanalyzer: fix rtio_log message extraction.

Fixes #550.
closed 2016-09-14 by whitequark