m-labs/artiq #551: Fire-and-forget RPC from the core device

jordens, 2016-09-03:
This addition allows RPCs from the core device which do not require a return value from the master/worker to return immediately, without incurring the cost of a network round trip plus latencies associated with the master/worker processes. This frees the core device kernel CPU to process further code, reducing experimental dead times. Core device RPCs for set_dataset() and mutate_dataset() shall be implemented as fire-and-forget RPCs.

Multiple fire-and-forget RPC requests may be queued in a FIFO manner. The runtime shall make data copies as appropriate to place in this queue to ensure that later mutations of the RPC arguments on the core device do not impact the RPC.

Regular (non fire-and-forget) RPCs shall wait for the fire-and-forget FIFO to empty before proceeding.
Mechanisms for handshaking between the kernel CPU and comms CPU shall be devised to enable data to be added to the FIFO queue and for the kernel CPU to return, even when the comms CPU is busy.
All such fire-and-forget RPCs shall be marked with a special Python decorator.

sbourdeauducq, 2016-09-14:
Let's call those "background RPCs" instead.

whitequark, 2016-10-30:
Needs testing:
  * interleaving of async and sync RPCs
  * exceptions in async RPCs

whitequark, 2016-11-01:
All done, tested, and documented.

closed 2016-11-01 by whitequark