m-labs/artiq #554: VADJ rail in doc

dhslichter, 2016-09-07:
We should probably specify that one needs to change the KC705 VADJ rail using PMBus in the "common problems" listing in the core_device.rst documentation file, instead of/in addition to having it where it currently is at the bottom of the QC2 documentation.

jordens, 2016-09-07:
That seems to imply that the QC2 documentation is usually not read or ignored when setting up a QC2 system. Is that the case?

sbourdeauducq, 2016-09-08:
Doesn't the "clock" design also require the same PMBus programming by the way? The doc currently does not say that.

dhslichter, 2016-09-08:
Yes, the point is that this is required for CLOCK and QC2, not just QC2, and so it should be mentioned in a place where people doing CLOCK also will look for it.  I don't know of anyone using KC705/QC1 currently -- I think we should move to deprecate it pending a check that nobody is using that combination.

sbourdeauducq, 2016-09-08:
What is the Penning lab using?

dhslichter, 2016-09-08:
HFGUI :P

dhslichter, 2016-09-08:
Penning confirms they are not using KC705/QC1 anymore. @jboulder are you using KC705/QC1 in any of your experiments these days?

sbourdeauducq, 2016-09-09:
What's their excuse?

closed 2016-09-09 by sbourdeauducq

dtcallcock, 2016-09-09:
They are too busy writing rubbish like this: http://science.sciencemag.org/content/352/6291/1297

jbqubit, 2016-10-04:
I missed that this was directed at me. I'm not using KC705/QC1. If Penning
moves to use ARTIQ I'd guess it will be after Sayma-Metlino become
available. Justin, thoughts? -Joe

-------
Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

dhslichter, 2016-10-04:
Justin has confirmed they are not using KC705/QC1, so it should be removed from the CI.

justboh, 2016-10-05:
For completeness, I will confirm we have no plans for QC1, and are setting up to use QC2 when we make the switch.

sbourdeauducq, 2016-10-05:
Do we drop AD9858 support as well or is someone using it (or planning to)?

sbourdeauducq, 2016-10-05:
That being said, if you want to give ARTIQ QC1 a quick test, it should work quite easily. Many of the first tests were done on it.

dhslichter, 2016-10-05:
I don't know of anyone planning to use AD9858.  That said, I don't think it is necessary to remove it from ARTIQ in case someone changes their mind.  It just won't be instantiated in any of the currently-compiled bitstreams.

commit 6150ab85 (2016-09-09): doc: move VADJ, closes #554
commit f23ebc03 (2016-09-09): doc: move VADJ, closes #554
commit 4ef5eb26 (2016-09-09): doc: move VADJ, closes #554