m-labs/artiq #556: Error if wrapper for rpc in kernel takes input of different types

r-srinivas, 2016-09-08:
Using 2.0.dev+563.gc4559fc on windows,

```python
from artiq.experiment import *

def print_type_x(x):
    print(type(x))

class RPC_Test(EnvExperiment):

    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        self.print_type_wrapper(1)
        self.print_type_wrapper(1.0)

    @kernel
    def print_type_wrapper(self, x):
        print_type_x(x)
```

I get the following error,

```
ERROR:worker(20316,rpc_test.py):root:While compiling <repository>\test\rpc_test.
py
<repository>\test\rpc_test.py:22:33-22:36: error: cannot unify float with numpy.
int?
        self.print_type_wrapper(1.0)
                                ^^^
<repository>\test\rpc_test.py:22:33-22:36: note: expression of type float
        self.print_type_wrapper(1.0)
                                ^^^
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 20316
```

If both functions get inputs of the same type, for example,

```python
from artiq.experiment import *

def print_type_x(x):
    print(type(x))

class RPC_Test(EnvExperiment):

    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        self.print_type_wrapper(1)
        self.print_type_wrapper(2)

    @kernel
    def print_type_wrapper(self, x):
        print_type_x(x)
```

It runs fine and I just get the output I expect,

```
INFO:worker(20317,rpc_test.py):print:<class 'numpy.int32'>
INFO:worker(20317,rpc_test.py):print:<class 'numpy.int32'>
```

If I take off the `@kernel` decorator for the function I also have no issues. 

```python
from artiq.experiment import *

def print_type_x(x):
    print(type(x))

class RPC_Test(EnvExperiment):

    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        self.print_type_wrapper(1)
        self.print_type_wrapper(1.0)

    def print_type_wrapper(self, x):
        print_type_x(x)
```

```
INFO:worker(20318,rpc_test.py):print:<class 'numpy.int32'>
INFO:worker(20318,rpc_test.py):print:<class 'float'>
```

sbourdeauducq, 2016-09-09:
Kernels must always be called with the same types.

closed 2016-09-09 by sbourdeauducq

r-srinivas, 2016-09-09:
Even in a case like this?

```python
from artiq.experiment import *

def print_type_x(x):
    print(type(x))

class RPC_Test(EnvExperiment):

    def build(self):
        self.setattr_device("core")
        self.setattr_argument("int", BooleanValue(True))

    @kernel
    def run(self):
        if self.int:
            self.print_type_wrapper(1)
        if not(self.int):
            self.print_type_wrapper(2.0)

    @kernel
    def print_type_wrapper(self, x):
        print_type_x(x)
```
That gives me a similar error.

sbourdeauducq, 2016-09-09:
Yes, because the compiler processes your whole run function (not just the branch that ends up executed), since it's a kernel.

r-srinivas, 2016-09-09:
Okay, good to know, thanks.