m-labs/artiq #557: If statements with strings in kernel

r-srinivas, 2016-09-09:
Using 2.0.dev+563.gc4559fc on windows, I have the following code,

```python
from artiq.experiment import *

class String_If(EnvExperiment):
  

    def build(self):
        self.setattr_device("core")
        self.x = 'test'

    @kernel
    def run(self):
        if self.x == 'test':
            print(True)
        else:
            print(False)
```

Which generates the following error message,

```
ERROR:worker(20396,string_if.py):root:Terminating with exception (OSError: Incor
rect reply from device: _D2HMsgType.LOAD_FAILED (expected _D2HMsgType.LOAD_COMPL
ETED))
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-08-31\lib\site-packages\artiq\master\worker
_impl.py", line 228, in main
    exp_inst.run()
  File "C:\Anaconda3\envs\artiq-2016-08-31\lib\site-packages\artiq\language\core
.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Anaconda3\envs\artiq-2016-08-31\lib\site-packages\artiq\coredevice\co
re.py", line 118, in run
    self.comm.load(kernel_library)
  File "C:\Anaconda3\envs\artiq-2016-08-31\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 299, in load
    self._read_empty(_D2HMsgType.LOAD_COMPLETED)
  File "C:\Anaconda3\envs\artiq-2016-08-31\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 138, in _read_empty
    self._read_expect(ty)
  File "C:\Anaconda3\envs\artiq-2016-08-31\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 134, in _read_expect
    format(self._read_type, ty))
OSError: Incorrect reply from device: _D2HMsgType.LOAD_FAILED (expected _D2HMsgT
ype.LOAD_COMPLETED)
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 20396
```

The problem seems to be having an if statement with strings in the kernel. Would it be possible to have support for them? For our particular case we can work around with integers but it could be useful to someone else in another situation.

If it's hard to support this, could it be specified in the documentation that if statements are with Boolean values or with integers/floats? Also, this error message seems a little hard to interpret.

whitequark, 2016-09-10:
This was already supported, just didn't work properly because of a separate bug.

closed 2016-09-10 by whitequark

r-srinivas, 2016-09-10:
That was quick, thanks!

commit b100770e (2016-09-10): runtime: expose strlen (implicitly required for comparing strings).

Fixes #557.
commit b73d7e40 (2016-09-10): runtime: expose strlen (implicitly required for comparing strings).

Fixes #557.