m-labs/artiq #562: DRTIO

jordens, 2016-09-26:
- https://github.com/m-labs/drtio_transceiver_test
- https://github.com/m-labs/artiq_drtio

sbourdeauducq, 2016-11-17:
Funded by Oxford (without switch support)

jbqubit, 2017-01-26:
What is the development checklist for DRTIO? How is is progressing?

jordens, 2017-01-26:
https://ssl.serverraum.org/lists-archive/artiq/2017-January/001083.html

jbqubit, 2017-01-26:
​The scope of DRTIO/DMA funded by Oxford remains obscure to me. ARL is also funding a big chunk of the distributed DMA. It would be helpful to know what subset of the features discussed in the wiki are expected under these combined contracts. My expectation based on past practice is a checklist. For example, 

https://github.com/m-labs/artiq/issues/580
https://github.com/m-labs/artiq/issues/561

Thoughts @hartytp?

sbourdeauducq, 2017-01-26:
@jbqubit As far as your contract goes, everything is in place for phase 2 except changing the frequency to match the DAC and more annoyingly fixing the obscure packet corruption bug (There is the sysref sync as well but this is not DRTIO). Also, this issue is not the best place for your message.

sbourdeauducq, 2017-02-04:
Packet corruption bug fixed.

sbourdeauducq, 2017-02-18:
@cjbe Basic DRTIO with outputs only works at 1.25Gbps with commit 9501d37378f2998bc250edd71de55526d2ee624c. You can have a look if you want.

sbourdeauducq, 2017-02-25:
Remaining steps:
- [ ] moninj
- [ ] get the Xilinx transceiver garbage to work over 1.25Gbps
- [ ] multiple satellites on one master
- [ ] inputs

sbourdeauducq, 2017-02-27:
Remote moninj working

sbourdeauducq, 2017-04-06:
There are still problems with:
* link initialization that fails intermittently, seems XTG-related so I will not spend time looking into it, maybe this just won't happen on GTH/Sinara.
* data rates other than 1.25Gbps, definitely XTG-related.
* ``core.reset()`` that causes spurious reports of satellite-side underflows, sequence errors and collisions, seems to be a logic bug.

Other than that, DRTIO systems with 2 boards (the master having local RTIO in addition to the satellite connection) have pretty much everything.

klickverbot, 2018-06-30:
Anything else missing from this umbrella bug?

sbourdeauducq, 2018-06-30:
No.

closed 2018-06-30 by sbourdeauducq