m-labs/artiq #566: ImportError for dashboard/browser with 2.0 on windows

r-srinivas, 2016-09-27:
Using 2.0 on windows I get the following error when trying to start the dashboard or the browser,

```
(artiq-2016-09-27) C:\Users\rabi>artiq_dashboard
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-09-27\Scripts\artiq_dashboard-script.py", l
ine 10, in <module>
    load_entry_point('artiq==2.0', 'gui_scripts', 'artiq_dashboard')()
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\setuptools-27.2.0-p
y3.5.egg\pkg_resources\__init__.py", line 565, in load_entry_point
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\setuptools-27.2.0-p
y3.5.egg\pkg_resources\__init__.py", line 2598, in load_entry_point
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\setuptools-27.2.0-p
y3.5.egg\pkg_resources\__init__.py", line 2258, in load
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\setuptools-27.2.0-p
y3.5.egg\pkg_resources\__init__.py", line 2264, in resolve
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\frontend\arti
q_dashboard.py", line 9, in <module>
    from PyQt5 import QtCore, QtGui, QtWidgets
ImportError: DLL load failed: The specified module could not be found.
```

Looking at the conda env,

```
(artiq-2016-09-27) C:\Users\rabi>conda list
# packages in environment at C:\Anaconda3\envs\artiq-2016-09-27:
#
Using Anaconda Cloud api site https://api.anaconda.org
alabaster                 0.7.9                    py35_0
babel                     2.3.4                    py35_0
cffi                      1.7.0                    py35_0
colorama                  0.3.7                    py35_0
docutils                  0.12                     py35_2
h5py                      2.6.0               np111py35_1
hdf5                      1.8.16                   vc14_0  [vc14]
icu                       57.1                     vc14_0  [vc14]
imagesize                 0.7.1                    py35_0
jinja2                    2.8                      py35_1
jpeg                      8d                       vc14_2  [vc14]
libpng                    1.6.22                   vc14_0  [vc14]
aiohttp                   0.17.2                     py_3    m-labs/label/main
artiq                     2.0             py_0+gita9bef3c    m-labs/label/main
asyncserial               0.1             py_11+git758c1e5    m-labs/label/main
binutils-or1k-linux       2.25.1                        4    m-labs/label/main
chardet                   2.2.1                      py_3    m-labs/label/main
dateutil                  2.4.2                      py_2    m-labs/label/main
levenshtein               0.12.0                   py35_1    m-labs/label/main
libgit2                   0.24.1                        7    m-labs/label/main
libssh2                   1.7.0.git                     5    m-labs/label/main
lit                       0.4.1                      py_2    m-labs/label/main
llvmlite-artiq            0.10.0.dev              py35_24    m-labs/label/main
outputcheck               0.4.2                      py_1    m-labs/label/main
prettytable               0.7.2                      py_3    m-labs/label/main
pygit2                    0.24.0                   py35_4    m-labs/label/main
pyqt5                     5.5.1                    py35_0    m-labs/label/main
pyqtgraph                 0.9.10          py_112+git7d34e22    m-labs/label/main

pythonparser              0.0                        py_6    m-labs/label/main
qt5                       5.5.1                         1    m-labs/label/main
quamash                   0.5.5                      py_1    m-labs/label/main
regex                     2015.11.22               py35_1    m-labs/label/main
sphinx-argparse           0.1.13                     py_4    m-labs/label/main
sphinxcontrib-wavedrom    1.1.0                      py_1    m-labs/label/main
markupsafe                0.23                     py35_2
mkl                       11.3.3                        1
numpy                     1.11.1                   py35_1
openssl                   1.0.2j                   vc14_0  [vc14]
pip                       8.1.2                    py35_0
pycparser                 2.14                     py35_1
pygit2                    0.24.0                    <pip>
pygments                  2.1.3                    py35_0
pyserial                  2.7                      py35_0
python                    3.5.2                         0
pytz                      2016.6.1                 py35_0
scipy                     0.18.1              np111py35_0
setuptools                27.2.0                   py35_1
sip                       4.18                     py35_0
six                       1.10.0                   py35_0
snowballstemmer           1.2.1                    py35_0
sphinx                    1.4.6                    py35_0
sphinx_rtd_theme          0.1.9                    py35_0
vs2015_runtime            14.0.25123                    0
wheel                     0.29.0                   py35_0
zlib                      1.2.8                    vc14_3  [vc14]
```

I didn't have this problem with my installation on 13/09 which uses the same version of qt so I'm not sure what changed.

r-srinivas, 2016-09-29:
Is this just a bug on my installation or do you see this also?

jordens, 2016-09-29:
Downgrade `icu` to what we have.
We'll need to pin it.

```patch
--- b   2016-09-29 20:42:49.673151669 +0200
+++ a   2016-09-29 20:42:49.641151106 +0200
@@ -1,25 +1,25 @@
-(artiq-2016-09-13) c:\artiq-magtrap>conda list
-# packages in environment at C:\Anaconda3\envs\artiq-2016-09-13:
+(artiq-2016-09-27) C:\Users\rabi>conda list
+# packages in environment at C:\Anaconda3\envs\artiq-2016-09-27:
 #
 Using Anaconda Cloud api site https://api.anaconda.org
-alabaster                 0.7.8                    py35_0
-babel                     2.3.3                    py35_0
+alabaster                 0.7.9                    py35_0
+babel                     2.3.4                    py35_0
 cffi                      1.7.0                    py35_0
 colorama                  0.3.7                    py35_0
 docutils                  0.12                     py35_2
 h5py                      2.6.0               np111py35_1
 hdf5                      1.8.16                   vc14_0  [vc14]
+icu                       57.1                     vc14_0  [vc14]
 imagesize                 0.7.1                    py35_0
 jinja2                    2.8                      py35_1
 jpeg                      8d                       vc14_2  [vc14]
 libpng                    1.6.22                   vc14_0  [vc14]
 aiohttp                   0.17.2                     py_3    m-labs/label/main
-artiq                     2.0rc1          py_0+git87fad87    m-labs/label/main
+artiq                     2.0             py_0+gita9bef3c    m-labs/label/main
 asyncserial               0.1             py_11+git758c1e5    m-labs/label/main
 binutils-or1k-linux       2.25.1                        4    m-labs/label/main
 chardet                   2.2.1                      py_3    m-labs/label/main
 dateutil                  2.4.2                      py_2    m-labs/label/main
-icu                       56.1                          0    m-labs/label/main
 levenshtein               0.12.0                   py35_1    m-labs/label/main
 libgit2                   0.24.1                        7    m-labs/label/main
 libssh2                   1.7.0.git                     5    m-labs/label/main
@@ -40,7 +40,7 @@
 markupsafe                0.23                     py35_2
 mkl                       11.3.3                        1
 numpy                     1.11.1                   py35_1
-openssl                   1.0.2h                   vc14_0  [vc14]
+openssl                   1.0.2j                   vc14_0  [vc14]
 pip                       8.1.2                    py35_0
 pycparser                 2.14                     py35_1
 pygit2                    0.24.0                    <pip>
@@ -48,13 +48,12 @@
 pyserial                  2.7                      py35_0
 python                    3.5.2                         0
 pytz                      2016.6.1                 py35_0
-requests                  2.11.1                   py35_0
-scipy                     0.18.0              np111py35_0
-setuptools                26.1.1                   py35_0
+scipy                     0.18.1              np111py35_0
+setuptools                27.2.0                   py35_1
 sip                       4.18                     py35_0
 six                       1.10.0                   py35_0
 snowballstemmer           1.2.1                    py35_0
-sphinx                    1.4.1                    py35_0
+sphinx                    1.4.6                    py35_0
 sphinx_rtd_theme          0.1.9                    py35_0
 vs2015_runtime            14.0.25123                    0
 wheel                     0.29.0                   py35_0
```

sbourdeauducq, 2016-10-02:
Works fine on win7-experimental. @r-srinivas can you try with a freshly installed conda?

r-srinivas, 2016-10-03:
Downgrading `icu` seems to fix it. If I try installing it it seems to install a higher version of `icu` from outside the m-labs channel for some reason.

```
(C:\Anaconda3) C:\Users\rabi>conda create -n artiq-2016-10-03 artiq
Using Anaconda Cloud api site https://api.anaconda.org
Fetching package metadata ...........
Solving package specifications: ..........

Package plan for installation in environment C:\Anaconda3\envs\artiq-2016-10-03

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    pytz-2016.7                |           py35_0         170 KB
    cffi-1.8.3                 |           py35_0         130 KB
    ------------------------------------------------------------
                                           Total:         300 KB

The following NEW packages will be INSTALLED:

    aiohttp:                0.17.2-py_3              m-labs/label/main
    alabaster:              0.7.9-py35_0
    artiq:                  2.0-py_0+gita9bef3c      m-labs/label/main
    asyncserial:            0.1-py_11+git758c1e5     m-labs/label/main
    babel:                  2.3.4-py35_0
    binutils-or1k-linux:    2.25.1-4                 m-labs/label/main
    cffi:                   1.8.3-py35_0
    chardet:                2.2.1-py_3               m-labs/label/main
    colorama:               0.3.7-py35_0
    dateutil:               2.4.2-py_2               m-labs/label/main
    docutils:               0.12-py35_2
    h5py:                   2.6.0-np111py35_1
    hdf5:                   1.8.16-vc14_0                              [vc14]
    icu:                    57.1-vc14_0                                [vc14]
    imagesize:              0.7.1-py35_0
    jinja2:                 2.8-py35_1
    jpeg:                   8d-vc14_2                                  [vc14]
    levenshtein:            0.12.0-py35_1            m-labs/label/main
    libgit2:                0.24.1-7                 m-labs/label/main
    libpng:                 1.6.22-vc14_0                              [vc14]
    libssh2:                1.7.0.git-5              m-labs/label/main
    lit:                    0.4.1-py_2               m-labs/label/main
    llvmlite-artiq:         0.10.0.dev-py35_24       m-labs/label/main
    markupsafe:             0.23-py35_2
    mkl:                    11.3.3-1
    numpy:                  1.11.1-py35_1
    openssl:                1.0.2j-vc14_0                              [vc14]
    outputcheck:            0.4.2-py_1               m-labs/label/main
    pip:                    8.1.2-py35_0
    prettytable:            0.7.2-py_3               m-labs/label/main
    pycparser:              2.14-py35_1
    pygit2:                 0.24.0-py35_4            m-labs/label/main
    pygments:               2.1.3-py35_0
    pyqt5:                  5.5.1-py35_0             m-labs/label/main
    pyqtgraph:              0.9.10-py_112+git7d34e22 m-labs/label/main
    pyserial:               2.7-py35_0
    python:                 3.5.2-0
    pythonparser:           0.0-py_6                 m-labs/label/main
    pytz:                   2016.7-py35_0
    qt5:                    5.5.1-1                  m-labs/label/main
    quamash:                0.5.5-py_1               m-labs/label/main
    regex:                  2015.11.22-py35_1        m-labs/label/main
    scipy:                  0.18.1-np111py35_0
    setuptools:             27.2.0-py35_1
    sip:                    4.18-py35_0
    six:                    1.10.0-py35_0
    snowballstemmer:        1.2.1-py35_0
    sphinx:                 1.4.6-py35_0
    sphinx-argparse:        0.1.13-py_4              m-labs/label/main
    sphinx_rtd_theme:       0.1.9-py35_0
    sphinxcontrib-wavedrom: 1.1.0-py_1               m-labs/label/main
    vs2015_runtime:         14.0.25123-0
    wheel:                  0.29.0-py35_0
    zlib:                   1.2.8-vc14_3                               [vc14]

Proceed ([y]/n)?
```

```
(C:\Anaconda3) C:\Users\rabi>conda info
Current conda install:

             platform : win-64
        conda version : 4.1.11
    conda-env version : 2.5.1
  conda-build version : 1.20.0
       python version : 3.5.1.final.0
     requests version : 2.9.1
     root environment : C:\Anaconda3  (writable)
  default environment : C:\Anaconda3
     envs directories : C:\Anaconda3\envs
        package cache : C:\Anaconda3\pkgs
         channel URLs : http://conda.anaconda.org/m-labs/label/main/win-64/
                        http://conda.anaconda.org/m-labs/label/main/noarch/
                        https://repo.continuum.io/pkgs/free/win-64/
                        https://repo.continuum.io/pkgs/free/noarch/
                        https://repo.continuum.io/pkgs/pro/win-64/
                        https://repo.continuum.io/pkgs/pro/noarch/
                        https://repo.continuum.io/pkgs/msys2/win-64/
                        https://repo.continuum.io/pkgs/msys2/noarch/
          config file : C:\Users\rabi\.condarc
         offline mode : False
    is foreign system : False
```

sbourdeauducq, 2016-10-04:
> a higher version of icu from outside the m-labs channel

Where from exactly?

r-srinivas, 2016-10-04:
I'm not sure. It's a little strange. I've specified the channel to install from and if I try to install only icu it's fine,

```
(C:\Anaconda3) C:\Users\rabi>conda create -n test icu -c m-labs
Using Anaconda Cloud api site https://api.anaconda.org
Fetching package metadata .............
Solving package specifications: ..........

Package plan for installation in environment C:\Anaconda3\envs\test:

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    icu-56.1                   |                0        11.1 MB  m-labs

The following NEW packages will be INSTALLED:

    icu: 56.1-0 m-labs

Proceed ([y]/n)? n
```

But if I try to install artiq it tries to grab the higher version

```
(C:\Anaconda3) C:\Users\rabi>conda create -n artiq-2016-10-04 artiq -c m-labs
Using Anaconda Cloud api site https://api.anaconda.org
Fetching package metadata .............
Solving package specifications: ..........

Package plan for installation in environment C:\Anaconda3\envs\artiq-2016-10-04:

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    binutils-or1k-linux-2.25.1 |                4        24.1 MB  m-labs
    libssh2-1.7.0.git          |                5         128 KB  m-labs
    chardet-2.2.1              |             py_3         445 KB  m-labs
    levenshtein-0.12.0         |           py35_1         107 KB  m-labs
    libgit2-0.24.1             |                7         604 KB  m-labs
    lit-0.4.1                  |             py_2         102 KB  m-labs
    llvmlite-artiq-0.10.0.dev  |          py35_24         6.2 MB  m-labs
    outputcheck-0.4.2          |             py_1         101 KB  m-labs
    prettytable-0.7.2          |             py_3          31 KB  m-labs
    pytz-2016.7                |           py35_0         170 KB
    regex-2015.11.22           |           py35_1         303 KB  m-labs
    aiohttp-0.17.2             |             py_3         108 KB  m-labs
    asyncserial-0.1            | py_11+git758c1e5           8 KB  m-labs
    cffi-1.8.3                 |           py35_0         130 KB
    dateutil-2.4.2             |             py_2         233 KB  m-labs
    pythonparser-0.0           |             py_6          42 KB  m-labs
    qt5-5.5.1                  |                1        34.9 MB  m-labs
    pygit2-0.24.0              |           py35_4         173 KB  m-labs
    pyqt5-5.5.1                |           py35_0         4.1 MB  m-labs
    pyqtgraph-0.9.10           |py_112+git7d34e22         431 KB  m-labs
    quamash-0.5.5              |             py_1          31 KB  m-labs
    sphinx-argparse-0.1.13     |             py_4          14 KB  m-labs
    sphinxcontrib-wavedrom-1.1.0|             py_1           5 KB  m-labs
    artiq-2.0                  |  py_0+gita9bef3c         1.1 MB  m-labs
    ------------------------------------------------------------
                                           Total:        73.5 MB

The following NEW packages will be INSTALLED:

    aiohttp:                0.17.2-py_3              m-labs
    alabaster:              0.7.9-py35_0
    artiq:                  2.0-py_0+gita9bef3c      m-labs
    asyncserial:            0.1-py_11+git758c1e5     m-labs
    babel:                  2.3.4-py35_0
    binutils-or1k-linux:    2.25.1-4                 m-labs
    cffi:                   1.8.3-py35_0
    chardet:                2.2.1-py_3               m-labs
    colorama:               0.3.7-py35_0
    dateutil:               2.4.2-py_2               m-labs
    docutils:               0.12-py35_2
    h5py:                   2.6.0-np111py35_1
    hdf5:                   1.8.16-vc14_0                   [vc14]
    icu:                    57.1-vc14_0                     [vc14]
    imagesize:              0.7.1-py35_0
    jinja2:                 2.8-py35_1
    jpeg:                   8d-vc14_2                       [vc14]
    levenshtein:            0.12.0-py35_1            m-labs
    libgit2:                0.24.1-7                 m-labs
    libpng:                 1.6.22-vc14_0                   [vc14]
    libssh2:                1.7.0.git-5              m-labs
    lit:                    0.4.1-py_2               m-labs
    llvmlite-artiq:         0.10.0.dev-py35_24       m-labs
    markupsafe:             0.23-py35_2
    mkl:                    11.3.3-1
    numpy:                  1.11.1-py35_1
    openssl:                1.0.2j-vc14_0                   [vc14]
    outputcheck:            0.4.2-py_1               m-labs
    pip:                    8.1.2-py35_0
    prettytable:            0.7.2-py_3               m-labs
    pycparser:              2.14-py35_1
    pygit2:                 0.24.0-py35_4            m-labs
    pygments:               2.1.3-py35_0
    pyqt5:                  5.5.1-py35_0             m-labs
    pyqtgraph:              0.9.10-py_112+git7d34e22 m-labs
    pyserial:               2.7-py35_0
    python:                 3.5.2-0
    pythonparser:           0.0-py_6                 m-labs
    pytz:                   2016.7-py35_0
    qt5:                    5.5.1-1                  m-labs
    quamash:                0.5.5-py_1               m-labs
    regex:                  2015.11.22-py35_1        m-labs
    scipy:                  0.18.1-np111py35_0
    setuptools:             27.2.0-py35_1
    sip:                    4.18-py35_0
    six:                    1.10.0-py35_0
    snowballstemmer:        1.2.1-py35_0
    sphinx:                 1.4.6-py35_0
    sphinx-argparse:        0.1.13-py_4              m-labs
    sphinx_rtd_theme:       0.1.9-py35_0
    sphinxcontrib-wavedrom: 1.1.0-py_1               m-labs
    vs2015_runtime:         14.0.25123-0
    wheel:                  0.29.0-py35_0
    zlib:                   1.2.8-vc14_3                    [vc14]

Proceed ([y]/n)?
```

e-jordan, 2016-10-13:
Hi there, 
I had exactly the same problem. Conda was installed two months ago (windows 7, 64bit). The artiq 1.3 gui was running. Now I upgraded to artiq 2.0, but the dashboard could not be opened. Threre was also icu 57.1 installed from [vc14]. 
[conda_artiq_list.txt](https://github.com/m-labs/artiq/files/528175/conda_artiq_list.txt)
After down grading icu to 56.1 the dashboard works. 
Thanks!

klickverbot, 2016-10-13:
Ah, I wish I had come across this earlier. I just came across this issue setting up a fresh conda environment (based on `artiq-kc705-nist_clock`) off the main channel. For the record, re-installing pyqt from pip also fixed the issue.

jordens, 2016-11-13:
https://github.com/m-labs/conda-recipes/commit/77efaef4b5e1129324de3cfbe5d494137311f

closed 2016-11-14 by jordens

r-srinivas, 2016-11-28:
This seemed to happen when I installed artiq 3.0 as well.

r-srinivas, 2017-02-07:
I ran into this trying to install artiq py_820+gitf4ae166. Also, it seems like the alabaster dependency was removed? At least I had to install that manually anyway.

cjbe, 2017-02-16:
I now also have this issue on several different machines with Artiq 2.2. Downgrading to icu 56.1 does not solve this, neither does re-installing pyqt from pip. Any suggestions?

jordens, 2017-02-16:
@r-srinivas @cjbe pretty sure you guys have a different issue. Please file a new one and provide information about what you have installed (context), what you did, and what your problem is (symptoms).