m-labs/artiq #568: Unable to connect to Pipistrello

vontell, 2016-09-28:
My apologies if this is the wrong place to post these kinds of issues.
I am attempting to connect to a Pipistrello board by following the instructions at [https://m-labs.hk/artiq/manual-release-2/installing.html#configuring-the-core-device](https://m-labs.hk/artiq/manual-release-2/installing.html#configuring-the-core-device). However, I am unsure of the exact specification for use in `device_db.pyon` (from the steps outlined at [https://m-labs.hk/artiq/manual-release-2/utilities.html#core-device-configuration-tool](https://m-labs.hk/artiq/manual-release-2/utilities.html#core-device-configuration-tool). When I use pppd, the output also always stops at the following:
```
Using interface ppp0
Connect: ppp0 <--> /dev/ttyUSB1
local  IP address 10.0.0.1
remote IP address 10.0.0.2
```
Also, when trying to connect using `flterm`, I get the following output (which repeats every five seconds or so in my terminal):
`~�}#�!}!}!} }4}"}&} } } } }%}&�J�I}'}"}(}"<�~`
Any help would be great!

sbourdeauducq, 2016-09-28:
Seems like everything is actually working. You can then connect to the board via TCP/IP.

vontell, 2016-09-28:
Thanks! I can see now that it is live, but I get a refused connection when attempting to connect (both through ssh or telnet):
`connect to host 10.0.0.2 port 22: Connection refused`
Is there additional setup that may be needed?

vontell, 2016-09-28:
Closing this as I believe these issues have been fixed. However, I was also wondering if there exists a database of examples for `device_db.pyon` files, specifically if anyone has one for the Pipistrello board. It would be nice to have this in an accessible location for ease of setup. (I have followed the tables and examples [here](https://m-labs.hk/artiq/manual-release-2/core_device.html#pipistrello), but with no luck in causing LEDs on my board to change).

closed 2016-09-28 by vontell

sbourdeauducq, 2016-09-29:
Why do you assume it runs telnet or SSH servers? Edit device_db.pyon to set the IP address of the board and the correct channel for the LED TTL, and you should be able to control it using the ARTIQ tools.

sbourdeauducq, 2016-09-29:
And the garbled characters on the flterm prompt are normal, they are the PPP initialization packets. You only see the human-readable text when the board is booting.

vontell, 2016-09-29:
That makes sense; thank you!