m-labs/artiq #569: Rust runtime roadmap

whitequark, 2016-10-01:
Needed for feature parity, with almost-only-Rust on comms CPU (there will still be `-lcompiler-rt`+`-lalloc` needed by Rust and `-lbase` needed by lwip):
 - [ ] Port watchdogs
 - [ ] Port cache
 - [ ] Port RPCs
 - [ ] Port startup kernels
 - [ ] Port idle kernels
 - [ ] Allow sessions to interrupt one another (magic_ok codepath)
 - [ ] Set SOF_KEEPALIVE on bind PCB
 - [ ] Add rustc --cfg support to misoc CSR generator
 - [ ] Port moninj
 - [ ] Port analyzer
 - [ ] Port isr/uart.h
 - [ ] Port/rewrite flash_storage

Needed for feature parity, with almost-only-Rust on kernel CPU:
 - [ ] Port DDS (to ARTIQ Python)
 - [ ] Port I2C (to ARTIQ Python?)
 - [ ] Port RTIO
 - [ ] Port artiq_personality

Needed for completeness:
 - [ ] TMPU support
 - [ ] Remove initial length field to allow streaming requests
 - [ ] Remove allocation when receiving kernel
 - [ ] Confine comms CPU stuff to bottom 4M of memory instead of having it all around

sbourdeauducq, 2016-10-01:
> Remove \0 from strings sent over the network

What is that for exactly?

whitequark, 2016-10-01:
Rust uses Pascal strings, not C strings, so that's just some junk that gets added on one end and removed on another.

whitequark, 2016-10-07:
> Remove allocation when receiving kernel

This turned out not to be possible because the shared objects we generate aren't execute-in-place (they have holes and that needs an MMU if we really want to do it zero-copy).

whitequark, 2016-10-07:
> Fully safe mailbox interface

I've evaluated this and, while possible, this conversion is unlikely to prevent more bugs than it would cause.

whitequark, 2017-02-03:
> Extract core_error
>  Fix problems with core_io and make it use core_error
>  Make byteorder use core_io and core_error

I've looked into it and core_io/core_error are very annoying to write, mostly because you have to manually produce patches for dozens of libstd versions. I'm not going to die on this hill, especially seeing as the workaround is mild. Someone else in the community is going to have to be annoyed enough.

whitequark, 2017-02-04:
Runtime has been migrated to Rust.

closed 2017-02-04 by whitequark