m-labs/artiq #571: scheduler.check_pause for graceful termination

r-srinivas, 2016-10-04:
I'm using artiq 2.0 on windows. It seems like you can use `scheduler.check_pause()` without having `scheduler.pause()`. This has the nice functionality of graceful termination in the kernel. For example,

```python
from artiq.experiment import *
from time import sleep

class Scheduler_Test(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("scheduler")

    @kernel
    def run(self):
        print("kernel starting")
        for i in range(100):
            if not(self.scheduler.check_pause()):
                print("running on kernel")
                sleep(1)
            else:
                print("terminated")
                break
```

Runs as per normal, and if a termination request or higher priority experiment is scheduled, it terminates gracefully. I'm not sure if this is a feature or a bug but it could be nice to include it in the documentation that this can be done. From the manual (https://m-labs.hk/artiq/manual-release-2/management_system.html?highlight=scheduler.check_pause#artiq.master.scheduler.Scheduler.check_pause) it doesn't seem clear that this can be called without `scheduler.pause`.

I'm not sure if there are any side effects of using `check_pause` like this but it seems to work as I described as far as I can tell.

jordens, 2016-10-05:
If it's a bug, there would be unexpected behavior. What's happening that's not expected?
IMHO the documentation of `check_pause()` is complete. It does exactly what it says it does. How you react to that is at your discretion. You could also do different things like actively kill jobs when `check_pause()`.
If you disagree, please suggest a sentence or two that make it clear that exactly this use case needs to be included in the documentation.

r-srinivas, 2016-10-05:
I guess from the source code pause isn't under the `scheduler` class in `artiq/master/scheduler.py`, but it could be useful to be listed in the Scheduler API Reference for completeness. 

For the description, I would add under Pauses

```
To check whether ``pause()`` would in fact *not* return immediately, use :meth:`artiq.master.scheduler.Scheduler.check_pause`. This method can be called even from a kernel.
```

For the `check_pause` description,

```
 """Returns ``True`` if there is a condition that could make ``pause``
        not return immediately (termination requested or higher priority run).
        The typical purpose of this function is to check from a kernel
        whether returning control to the host and pausing would have an effect,
        in order to avoid the cost of switching kernels in the common case
        where ``pause`` does nothing. This function can also be called without
        ``pause`` to allow termination of a kernel in a predefined safe state.
        """
```

Mainly, from the description it seemed that `check_pause` needs to be called with `pause` but that isn't the case. It would be useful for users to be aware of that. An example experiment that uses this could also be handy.

commit 69099691 (2016-10-17): doc: clarify usage of pause/check_pause, closes #571
commit 5ecfa819 (2016-10-17): doc: clarify usage of pause/check_pause, closes #571
closed 2016-10-17 by sbourdeauducq