m-labs/artiq #573: Meta-experiments in artiq

r-srinivas, 2016-10-05:
I want to write an experiment that runs some other experiment that sets some hardware based on a parameter x, and then runs another experiment that generates a fit whose value is y. I then want to plot y v x.  I'm having an issue where I need to add a delay after `self.scheduler.pause()` in order to ensure that the `analyze` of an experiment runs before the next experiment is scheduled. 

```python
from artiq.experiment import *
from time import time, sleep

class Experiment_A(EnvExperiment):
    
    def run(self):
        print("Ran experiment A.")

    def analyze(self):
        print("Analysed experiment A")

class Scheduler_Submit_Test_2(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("scheduler")
        
    def run(self):
        for i in range(2):
            print("Scheduled experiment A, i={}.".format(i))
            self.scheduler.submit(
                    pipeline_name ='main',
                    priority=1,
                    expid = {'class_name': 'Experiment_A',
                             'repo_rev': 'N/A',
                             'file': 'test\\scheduler_submit_test.py',
                             'log_level': 30, 'arguments': {}},
                    due_date = time(),
                    flush=False)
            sleep(5)
            self.scheduler.pause()
            sleep(5)
```

This produces, 

```
INFO:worker(20424,scheduler_submit_test.py):print:Scheduled experiment A, i=0.
INFO:worker(20425,scheduler_submit_test.py):print:Ran experiment A.
INFO:worker(20425,scheduler_submit_test.py):print:Analysed experiment A
INFO:worker(20424,scheduler_submit_test.py):print:Scheduled experiment A, i=1.
INFO:worker(20426,scheduler_submit_test.py):print:Ran experiment A.
INFO:worker(20426,scheduler_submit_test.py):print:Analysed experiment A
```

Which is what I want. However, if I take away the last delay, I get

```
INFO:worker(20427,scheduler_submit_test.py):print:Scheduled experiment A, i=0.
INFO:worker(20428,scheduler_submit_test.py):print:Ran experiment A.
INFO:worker(20427,scheduler_submit_test.py):print:Scheduled experiment A, i=1.
INFO:worker(20428,scheduler_submit_test.py):print:Analysed experiment A
INFO:worker(20429,scheduler_submit_test.py):print:Ran experiment A.
INFO:worker(20429,scheduler_submit_test.py):print:Analysed experiment A
```

Where the second experiment is scheduled before the first one is analysed. Is there a way to accomplish this without a delay? It seems like flush isn't an answer, as if I set that to be `True`, I get

```
INFO:worker(20430,scheduler_submit_test.py):print:Scheduled experiment A, i=0.
INFO:worker(20430,scheduler_submit_test.py):print:Scheduled experiment A, i=1.
INFO:worker(20431,scheduler_submit_test.py):print:Ran experiment A.
INFO:worker(20431,scheduler_submit_test.py):print:Analysed experiment A
INFO:worker(20432,scheduler_submit_test.py):print:Ran experiment A.
INFO:worker(20432,scheduler_submit_test.py):print:Analysed experiment A
```

As `Scheduler_Submit_Test_2` was submitted earlier, it gets flushed before `Experiment_A` runs even though `Experiment_A` is at a higher priority. Is this expected behaviour or should flush only apply to experiments at the same or higher priority?

sbourdeauducq, 2016-10-05:
Why do you use the scheduler for that - why not import the code of the sub experiment you want and run it directly in the same worker by calling its methods?
If you want to avoid the double entry in the explorer, do this: ``import OtherExperiment as _OtherExperiment``.

sbourdeauducq, 2016-10-05:
If you do need to use the scheduler, I believe that the following should do what you want:
```python
            self.scheduler.submit(
                    expid = {'class_name': 'FirstExperiment',
                             'repo_rev': 'N/A',
                             'file': 'test\\scheduler_submit_test.py',
                             'log_level': 30, 'arguments': {}})
            self.scheduler.submit(
                    expid = {'class_name': 'SecondExperimentDependingOnFirst',
                             'repo_rev': 'N/A',
                             'file': 'test\\scheduler_submit_test.py',
                             'log_level': 30, 'arguments': {}},
                    flush=True)
```

jordens, 2016-10-05:
Sub-experiments have different and -- depending on the use case -- unacceptable semantics w.r.t. results.

Have a look at the ipython notebook example. It demonstrates how you could poll for experiment completion. A notification API for this would be a nice feature though.

r-srinivas, 2016-10-05:
Thanks, the ipython notebook was helpful. I'll look into using subexperiments but it seemed like that would have other issues. For example, if that has some gui arguments that are used I don't want that to necessarily present itself in the meta experiment, and I'd want the meta experiment to set some of the arguments which would maybe require some restructuring of the experiment code.

I'd agree, some sort of function that returns when an experiment is done, something like `scheduler.complete(rid)' or something would be handy.

closed 2016-10-05 by jordens

sbourdeauducq, 2016-10-05:
> if that has some gui arguments that are used I don't want that to necessarily present itself in the meta experiment

That will take a bit of digging into the ARTIQ sources, but you can remove the arguments of the subexperiment from the GUI and set them from the top-level experiment.