m-labs/artiq #574: moninj is broken

whitequark, 2016-10-05:
Setting OE=1 doesn't seem to do anything (in particular OE remains at 0 as read by monitor).

sbourdeauducq, 2016-11-04:
@whitequark any progress on this?

whitequark, 2016-11-04:
No. I have no idea where even to start.

cjbe, 2016-12-21:
I think I have seen this issue too: The injection monitoring does not work until rtio_init() has been called - this is only observable if one does not have a startup kernel that calls core.reset().

cjbe, 2016-12-21:
Sure - this is just a gotcha when you are setting new hardware up: If you flash the board, and the first thing you try to do is toggle a TTL using the override it does not work.

sbourdeauducq, 2016-12-21:
@whitequark did you see a problem specifically on OE, or was moninj simply not responding at all (until there was at least one core reset)?

whitequark, 2016-12-21:
@sbourdeauducq it was definitely responding because I've seen some other pins at 1

sbourdeauducq, 2017-02-27:
Works fine as per my test. Sending injection commands on OE does cause the device to send OE change notifications on a monitored channel. @whitequark maybe you used output-only (TTLOut) channels? Those do not have a OE probe or injector.

closed 2017-02-27 by sbourdeauducq

whitequark, 2017-02-27:
> maybe you used output-only (TTLOut) channels? Those do not have a OE probe or injector.

I've definitely not used output-only channels. Not only I specifically looked them up in ddb but for good measure I verified every single one anyway.