m-labs/artiq #576: remove ad9858 and kc7005-nist_qc1 support

jordens, 2016-10-05:
#554 
Unless there is protest.

sbourdeauducq, 2016-10-05:
We can simply not port AD9858 to the new Rust/ARTIQ-Python implementation (@whitequark).
I suppose the Pipistrello would be TTL-only then?

dhslichter, 2016-10-05:
Having the Pipistrello be TTL-only seems good.  A few comments on this front:
- Pipistrello being the initial demo system that many groups use, having a few more TTLs is handy.
- I would suggest using the lines formerly taken up by the DDS bus for a combination of TTLInOut and SPI channels.  
- We have written an SPI driver for the AD9914; once complete, we will submit as a patch.  This will enable people trying out ARTIQ to run with Pipistrello + AD9914 eval board(s) programmed over SPI.  Modifications of the AD9914 SPI driver would be simple, and could be made by users desiring to use a different DDS over SPI via an eval board.

commit 5e8888d5 (2016-11-21): Fully drop AD9858 and kc705-nist_qc1 support (closes #576).
closed 2016-11-21 by whitequark