m-labs/artiq #580: [meta] phaser: jesd204b dds demo

jordens, 2016-10-05:
https://github.com/m-labs/artiq/tree/phaser

- [x] parallelized CORDIC
- [x] CORDIC tests
- [x] parallelized phase accumulator
- [x] PhasedAccu tests
- [x] RTIO channels for amplitude, frequency, phase
- [x] test RTIO channels
- [x] kernel support for SAWG RTIO channels
- [x] document SAWG RTIO channels
- [x] test kernel support
- [x] SPI support for AD9154-FMC-EBZ: AD9516 and AD9154
- [x] test SPI support
- [x] Register maps for both AD9516, AD9145
- [x] verify register maps
- [x] document SPI and register maps
- [x] clock tree SPI configuration with internal RTIO clock bootstrap for startup_kernel
- [x] external RTIO clock
- [x] measure/verify clocks
- [x] measure/verify SYSREF
- [x] DAC clocking SPI configuration
- [x] DAC SERDES PLL lock
- [x] write complete DAC SPI configuration, JESD, datapath, phy, transport, sync
- [x] document installation, setup, clocking, usage
- [ ] finish JESD204B core (https://github.com/enjoy-digital/litejesd204b/TODO
- [ ] hook up SAWG outputs to JESD204B core
- [ ] test complete

jordens, 2016-10-14:
[image]

jordens, 2016-10-14:
[image]

* 4 channels (two shown)
* 10 GHz line rate (out of spec)
* 500 MHz data rate
* 4x interpolation to 2 GHz sample rate
* executing [demo.py](https://github.com/m-labs/artiq/blob/4b4fd32e3d756e555c593ba33a621c6602b32e5c/artiq/examples/phaser/repository/demo.py)

closed 2016-10-14 by jordens

jbqubit, 2016-10-14:
Please wait to see if I can reproduce on UMD hardware before closing.

jordens, 2016-10-14:
Please file new issues when and where you find them.