m-labs/artiq #581: input-only and differential TTLs

jordens, 2016-10-05:
At least on Xilinx, an input-only pin have much wider IOStandard compatibility (you can have an LVDS_25 input on a VCCIO=3.3 bank but not an output).

Cleanup (make this less of a hack) https://github.com/m-labs/artiq/commit/a91ed8394c2964bc335747f92a0ece5486b8009a and take: https://github.com/m-labs/artiq/commit/279f0d568dd96dc4966ebc6862cd2ec0562d742b

sbourdeauducq, 2016-10-06:
Do the Xilinx tools complain when you use the regular I+O PHY?

jordens, 2016-10-06:
Yes. They don't allow it.

sbourdeauducq, 2016-10-17:
Is that really necessary considering that sysref sampling will use a more precise IDELAY scan eventually?

jordens, 2016-10-17:
It's not required for sysref. But it's a feature. And if somebody wants/needs it this is how to do it. Leave it around.

jordens, 2017-08-28:
differential channels were implemented 95c885b580a8d09860c39fe54d82a0a1ab915ed9