m-labs/artiq #582: plan ARTIQ users group meeting

jbqubit, 2016-10-10:
An in-person ARTIQ users group meeting would be beneficial for current and prospective ARTIQ users. This Issue is meant to get a conversation started on this topic. 

# When and where? 
* Before or after an existing physics conference would be convenient. Ideally, a conference of interest to the existing ARTIQ user community plus scientists from other control-limited disciplines. 
    * APS DAMOP 2017 is June 5-9 in Sacramento, CA
    * APS March Meeting 2017 is March 13-17 in New Orleans, LA
    * What are candidate European conferences? 
* Location should be near an international airport for ease of travel by m-labs and European users
* Should there be separate European and American meetings?

# Administrative Roles
A conference needs organizers of 
* publicity
* logistics (fee collection, venue, housing, bulletin, website)
* curriculum for ARTIQ school (eg graduate students)
* conference program
* breakout sessions
* social activities

# Questions for ARTIQ users
* What would you like to see at such a conference?
* Would you attend a conference if it were on your continent? Another continent?

jordens, 2016-10-11:
@jboulder let's not use this issue tracker for brainstorming and note-taking. The mailing list is a better place.

jbqubit, 2016-12-16:
@jordens please reopen this.

sbourdeauducq, 2017-01-02:
Really, let's use the mailing list for this.

closed 2017-01-02 by sbourdeauducq