m-labs/artiq #587: hd5 file does not contain all data used

r-srinivas, 2016-10-14:
Using artiq 2.0 on windows, I ran into a key error when trying to load an experiment to reanalyse,

```
root:Terminating with exception (KeyError: "'mw.f.3p3_2p2'")
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\master\worker.py", line 219, in _handle_worker_requests
    data = func(*obj["args"], **obj["kwargs"])
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\browser\experiments.py", line 385, in get
    return self._data.backing_store[key][1]
KeyError: "'mw.f.3p3_2p2'"
```
This is is a datum that is used to set the default argument but isn't set by the experiment. In more detail, if we set the value in the dataset `self.set_dataset("data.browser.x", 2, persist=True)`, 

and use this to instantiate a value in an experiment,

```python
from artiq.experiment import *

class Browser_Test(EnvExperiment):

    def build(self):
        self.setattr_device("core")
        self.setattr_argument("x",
                              NumberValue(self.get_dataset("data.browser.x")))

    @kernel
    def run(self):
        print(self.x)

    def analyze(self):
        self.set_dataset("data.browser.test.x", self.x,
                         broadcast=True,
                         save=True)
```

I get this error when trying to open the experiment in the browser.
```
root:Terminating with exception (KeyError: "'data.browser.x'")
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\master\worker.py", line 219, in _handle_worker_requests
    data = func(*obj["args"], **obj["kwargs"])
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\browser\experiments.py", line 385, in get
    return self._data.backing_store[key][1]
KeyError: "'data.browser.x'"
```

Am I missing something? If not, one solution would be to save all the values from the dataset that are called in anyway by an experiment (through `get_dataset`). Saving all persistent dataset values to every hd5 file could be another, possibly better solution if we want to keep track of all ion parameters very experiment.

sbourdeauducq, 2016-10-17:
ARTIQ doesn't do anything there by default, because the values of the input datasets can change while the experiment is executing (think of a long experiment that pauses, and gets interrupted by periodic recalibrations).
Because of this possible change of values, snapshots of all persistent values in every HDF5 is not the right solution for keeping track of ion parameters - the influxdb logger is more appropriate, or analyzing multiple HDF5s (since the experiments producing changes will also generate their own output file).
For those input datasets that you know will not change, you can save them manually in the HDF5 with ``set_dataset(save=True)``. I'm OK with building this functionality into ARTIQ, by adding an ``archive`` argument to ``get_dataset`` that also enforces that you can get each dataset only once.

r-srinivas, 2016-10-17:
>I'm OK with building this functionality into ARTIQ, by adding an archive argument to get_dataset that also enforces that you can get each dataset only once.

I think that's fine for us, except for meta experiments like this.  I guess we can pass an index maybe to an experiment but seems slightly cleaner this way.

```python
from artiq.experiment import *
from time import sleep, time
import numpy as np

class MW_RSB_Frequency_Drift_Scan(EnvExperiment):
    '''MW RSB Frequency Drift Scan'''

    def build(self):
        self.setattr_device("core")
        self.setattr_device("scheduler")
        self.setattr_argument("time_min",
                              Scannable(default=LinearScan(0, 10, 11)))

    def prepare(self):
        self.x_list = list(self.time_min)
        # test
        l = len(self.x_list)
        self.delay = ((self.x_list[-1]-self.x_list[0])/(l-1))*60  # min to s
        self.xvar = "data.mw.rsb.drift"
        self.xlabel = self.xvar + ".x"
        self.ylabel = self.xvar + ".y"
        self.set_dataset(self.xlabel, np.full(l, np.nan),
                         broadcast=True, save=True)
        self.set_dataset(self.ylabel, np.full(l, np.nan),
                         broadcast=True, save=True)
        print(l)
        print(self.delay)

    def run(self):
        i = 0
        start = time()
        for x in self.x_list:
            exp_start = time()
            expid = dict(
                repo_rev='N/A',
                file='microwaves\\microwave_rsb_scan.py',
                class_name='Microwave_RSB_Scan',
                log_level=30,
                arguments=dict(
                    mw_dds_frequency=(self.get_dataset("mw.f.3p1_2p1") +
                                      self.get_dataset("mw.d.3p1_2p1")),
                    radial_f=self.get_dataset("trap.radial_1.mw.f"),
                    offset=dict(ty='LinearScan',
                                stop=5000,
                                start=-5000,
                                npoints=100),
                    t_pulse=dict(ty='NoScan',
                                 repetitions=1,
                                 value=0.0015),
                    p_2=self.get_dataset("mw.nulling.p_2"),
                    a_2=self.get_dataset("mw.nulling.a_2"),
                    p_3=0,
                    a_3=0.9,
                    rsb=False,
                    gs_cool=False,
                    f_detect=330*MHz,
                    t_detect=0.0001,
                    repetitions=100,
                    nbins=100,
                    cold=False,
                    reference_histogram='Use manual detection parameters',
                    save_fit=True,
                    drift=True))
            rid = self.scheduler.submit(pipeline_name='main',
                                        expid=expid,
                                        priority=1,
                                        due_date=None,
                                        flush=False)
            
            # wait for experiments to finish before fitting
            while rid in self.scheduler.get_status():
                sleep(0.1)
                self.scheduler.pause()
            core_times = self.get_dataset("data.mw.rsb.drift.core_time")
            frequencies = self.get_dataset("data.mw.rsb.drift.frequency")
            frequency = self.get_dataset("trap.radial_1.mw.f")
            x = ((frequency-frequencies[0])/(frequencies[1] - frequencies[0]) *
                 (core_times[1]-core_times[0])) + core_times[0]
            self.mutate_dataset(self.xlabel, i, x)
            self.mutate_dataset(self.ylabel, i, frequency)
            i += 1
            if i < len(self.x_list):
                while time()-exp_start < self.delay:
                    sleep(5)
            ion_counts = self.get_dataset("data.im_counts")
            if ion_counts < 10:
                print("Lost ion.")
                break
        print(time()-start)
```

sbourdeauducq, 2016-10-18:
> while time()-exp_start < self.delay:
>                     sleep(5)

``sleep(self.delay-(time()-exp_start))`` ?

sbourdeauducq, 2016-10-18:
> core_times = self.get_dataset("data.mw.rsb.drift.core_time")
> frequencies = self.get_dataset("data.mw.rsb.drift.frequency")
> frequency = self.get_dataset("trap.radial_1.mw.f")

I would locate and open the HDF5 file corresponding to ``rid``. Using broadcasted datasets for getting values from the sub-experiments is not robust as the datasets may have been modified in the meantime.

sbourdeauducq, 2016-10-18:
And it is not a problem anyway for the HDF5 archival proposal since you could simply set ``archive=False`` on those dataset requests and issue them multiple times.

commit 69d96b01 (2016-10-18): master: archive input datasets. Closes #587
closed 2016-10-18 by sbourdeauducq