m-labs/artiq #588: phaser build: No rule to make target `divsi3.o', needed by `libcompiler-rt.a'

jbqubit, 2016-10-16:
Following these instructions
```
https://github.com/m-labs/artiq/blob/phaser/README_PHASER.rst
```

```
(artiq-phaser) britton@britton1:~/artiq-dev-phaser/artiq$ python -m artiq.gateware.targets.kc705 -H phaser --toolchain vivado
make: Entering directory `/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libcompiler-rt'
make: *** No rule to make target `divsi3.o', needed by `libcompiler-rt.a'.  Stop.
make: Leaving directory `/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libcompiler-rt'
Traceback (most recent call last):
  File "/home/britton/anaconda3/envs/artiq-phaser/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/britton/anaconda3/envs/artiq-phaser/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/britton/artiq-dev-phaser/artiq/artiq/gateware/targets/kc705.py", line 602, in <module>
    main()
  File "/home/britton/artiq-dev-phaser/artiq/artiq/gateware/targets/kc705.py", line 598, in main
    build_artiq_soc(soc, builder_argdict(args))
  File "/home/britton/artiq-dev-phaser/artiq/artiq/gateware/soc.py", line 59, in build_artiq_soc
    builder.build()
  File "/home/britton/artiq-dev/misoc/misoc/integration/builder.py", line 157, in build
    self._generate_software()
  File "/home/britton/artiq-dev/misoc/misoc/integration/builder.py", line 134, in _generate_software
    subprocess.check_call(cmd)
  File "/home/britton/anaconda3/envs/artiq-phaser/lib/python3.5/subprocess.py", line 581, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libcompiler-rt']' returned non-zero exit status 2
```

whitequark, 2016-10-17:
Have you checked out the submodules?

jordens, 2016-10-17:
You did not install misoc correctly (or not as per the artiq instructions).

jbqubit, 2016-10-18:
Ah. Yes. I didn't know there were new submodules to snag. Not a bug. 
```
$ git pull --recurse-submodules
```

closed 2016-10-18 by jbqubit

sbourdeauducq, 2016-10-18:
Not new, they'be been there since the beginning of artiq.