m-labs/artiq #590: [phaser] ksupport.c:412:55: error: no member named 'init' in 'struct dyld_info'

jordens, 2016-10-18:
Another one...

```
(artiq-phaser) britton@britton1:~/artiq-dev-phaser/artiq$ python -m
artiq.gateware.targets.kc705 -H phaser --toolchain vivado
make: Entering directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libcompiler-rt'
make: Nothing to be done for `all'.
make: Leaving directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libcompiler-rt'
make: Entering directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libbase'
make: Nothing to be done for `all'.
make: Leaving directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libbase'
make: Entering directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libnet'
make: Nothing to be done for `all'.
make: Leaving directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libnet'
make: Entering directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/bios'
 CC       main.o
 LD       bios.elf
chmod -x bios.elf
 OBJCOPY  bios.bin
chmod -x bios.bin
python3 -m misoc.tools.mkmscimg bios.bin
make: Leaving directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/bios'
make: Entering directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/liballoc'
make: Nothing to be done for `all'.
make: Leaving directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/liballoc'
make: Entering directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libm'
make: Nothing to be done for `all'.
make: Leaving directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libm'
make: Entering directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libdyld'
make: Nothing to be done for `all'.
make: Leaving directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libdyld'
make: Entering directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libunwind'
make: Nothing to be done for `all'.
make: Leaving directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/libunwind'
make: Entering directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/liblwip'
rm -f lwipopts.h
rm -f arch
ln -s /home/britton/artiq-dev-phaser/artiq/artiq/runtime/liblwip/lwipopts.h
lwipopts.h
ln -s /home/britton/artiq-dev-phaser/artiq/artiq/runtime/liblwip/arch arch
mkdir -p core/ipv4
mkdir -p netif
mkdir -p netif/ppp
make: Leaving directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/liblwip'
make: Entering directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/runtime'
 CC       ksupport.o
/home/britton/artiq-dev-phaser/artiq/artiq/runtime/ksupport.c:401:40:
error: too few arguments to
      function call, expected 6, have 5
                      &load_reply.error)) {
                                       ^
/home/britton/artiq-dev/misoc/misoc/software/include/dyld/dyld.h:22:1:
note: 'dyld_load' declared here
int dyld_load(const void *shlib, Elf32_Addr base,
^
/home/britton/artiq-dev-phaser/artiq/artiq/runtime/ksupport.c:412:55:
error: no member named 'init' in
      'struct dyld_info'
        void (*kernel_run)() = request->library_info->init;
                               ~~~~~~~~~~~~~~~~~~~~~  ^
2 errors generated.
make: *** [ksupport.o] Error 1
make: Leaving directory
`/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/runtime'
Traceback (most recent call last):
  File "/home/britton/anaconda3/envs/artiq-phaser/lib/python3.5/runpy.py",
line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/britton/anaconda3/envs/artiq-phaser/lib/python3.5/runpy.py",
line 85, in _run_code
    exec(code, run_globals)
  File "/home/britton/artiq-dev-phaser/artiq/artiq/gateware/targets/kc705.py",
line 602, in <module>
    main()
  File "/home/britton/artiq-dev-phaser/artiq/artiq/gateware/targets/kc705.py",
line 598, in main
    build_artiq_soc(soc, builder_argdict(args))
  File "/home/britton/artiq-dev-phaser/artiq/artiq/gateware/soc.py",
line 59, in build_artiq_soc
    builder.build()
  File "/home/britton/artiq-dev/misoc/misoc/integration/builder.py",
line 157, in build
    self._generate_software()
  File "/home/britton/artiq-dev/misoc/misoc/integration/builder.py",
line 134, in _generate_software
    subprocess.check_call(cmd)
  File "/home/britton/anaconda3/envs/artiq-phaser/lib/python3.5/subprocess.py",
line 581, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C',
'/home/britton/artiq-dev-phaser/artiq/misoc_phaser_kc705/software/runtime']'
returned non-zero exit status 2
```

whitequark, 2016-10-18:
please update the submodules.

jordens, 2016-10-18:
> `/home/britton/artiq-dev/misoc/misoc/software/include/dyld/dyld.h:22:1:`

Have you shot your installation? Do you have misoc in `artiq-dev` and `artiq-phaser-dev`?

whitequark, 2016-10-18:
correction: please apply this patch:
```diff
diff --git a/artiq/runtime/ksupport.c b/artiq/runtime/ksupport.c
index 4bc8abe..8236166 100644
--- a/artiq/runtime/ksupport.c
+++ b/artiq/runtime/ksupport.c
@@ -409,7 +409,7 @@ int main(void)
     }
 
     if(request->run_kernel) {
-        void (*kernel_run)() = request->library_info->init;
+        void (*kernel_run)() = dyld_lookup("__modinit__", request->library_info);
         void *typeinfo = dyld_lookup("typeinfo", request->library_info);
 
         mailbox_send_and_wait(&load_reply);
```

jordens, 2016-10-18:
@whitequark should that patch go into the phaser branch?

whitequark, 2016-10-18:
@jordens maybe just merge master into it?

commit 54f05b6f (2016-10-18): ksupport: kernel_run lookup with dyld (closes #590)

adapt to misoc change
closed 2016-10-18 by jordens