m-labs/artiq #591: run another instcombine+sccp after licm or something like that

jordens, 2016-10-18:
```
10:11 < rjo> whitequark: a quick question. https://github.com/m-labs/artiq/blob/phaser/artiq/examples/phaser/repository/demo.py even with kernel_invariants and
             fast-math, this generates lots of muldf3 and fixdfsi. anything obvious that prevents that from being const-folded?
10:12 < whitequark> set_phase and set_amplitude are syscalls, right?
10:12 < whitequark> actually, no, nevermind
10:12 < whitequark> do ARTIQ_DUMP_LLVM=foo.ll artiq_run demo.py
10:12 < whitequark> and give me the file
10:13 < rjo> http://hastebin.com/fuguwemuqa.pl
10:15 < rjo> dropping the round() does not help
10:15 < whitequark> hmmmm
10:16 < rjo> it's not urgent at all. just something i could not understand.
10:16 < whitequark> no, that's wrong
10:16 < whitequark> the round intrinsic is marked as safe to speculatively execute
10:16 < whitequark> I did that in April
10:17 < whitequark> oh, I see
10:17 < whitequark> >curl http://hastebin.com/raw/fuguwemuqa | opt -std-link-opts -S
10:17 < whitequark> observe the output
10:18 < whitequark> we need to run another instcombine+sccp after licm or something like that
10:19 < whitequark> file an issue, I'll fix it later. it's not hard, just needs some time to figure out why the existing pipeline isn't enough
 ```

http://hastebin.com/fuguwemuqa.ll

whitequark, 2016-11-14:
This actually already doesn't have the problem on the master/phaser branch. I believe the recent fixes for LLVM performance also addressed this issue.

sbourdeauducq, 2016-11-14:
Is that fixed in release-2?

whitequark, 2016-11-14:
No, since release-2 doesn't run the testcase...

sbourdeauducq, 2016-11-14:
Well, @jordens added this to the 2.1 milestone. Should it be moved to 3.0 then?

jordens, 2016-11-14:
If it also happens to e.g. the DDS code in 2.0, which is likely, and if it is possible to backport, I'd like to see this fixed in 2.1.

jordens, 2016-11-20:
@whitequark Is this more difficult than adding a few more passes?

whitequark, 2016-11-20:
@jordens The fix in the phaser branch was completely different, in fact it didn't involve adding passes at all (but making the analyses more precise); this requires migrating 2.x to LLVM 3.9.

I could add the passes as described above, but, in light of the way this was fixed in master, this seems sloppy, and I would really like to investigate first with some representative code (that runs on 2.x) whether this is the right solution, or whether there is some other bug.

whitequark, 2016-11-21:
Reopening as per IRC discussion.

whitequark, 2017-04-12:
Closing as per discussion with @sbourdeauducq, in favor of asking dowstream to upgrade to 3.0.

closed 2017-04-12 by whitequark

jordens, 2017-04-12:
Ok by me. Just need to verify that it does not occur in master.