m-labs/artiq #592: phaser: RuntimeError: LLVM IR parsing error

jbqubit, 2016-10-18:
$ artiq_run repository/dac_setup.py
====== LLVM IR (BROKEN) DUMP ======
LLVM IR (broken) dumped as /tmp/tmp3hbuka_t.ll
Traceback (most recent call last):
  File "/home/britton/anaconda3/envs/artiq-phaser/bin/artiq_run", line 11, in <module>
    load_entry_point('artiq==3.0.dev0+159.gbf942fc', 'console_scripts', 'artiq_run')()
  File "/home/britton/artiq-dev-phaser/artiq/artiq/frontend/artiq_run.py", line 221, in main
    return run(with_file=True)
  File "/home/britton/artiq-dev-phaser/artiq/artiq/frontend/artiq_run.py", line 207, in run
    raise exn
  File "/home/britton/artiq-dev-phaser/artiq/artiq/frontend/artiq_run.py", line 200, in run
    exp_inst.run()
  File "/home/britton/artiq-dev-phaser/artiq/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/core.py", line 111, in run
    self.compile(function, args, kwargs, set_result)
  File "/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/core.py", line 95, in compile
    library = target.compile_and_link([module])
  File "/home/britton/artiq-dev-phaser/artiq/artiq/compiler/targets.py", line 180, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules],
  File "/home/britton/artiq-dev-phaser/artiq/artiq/compiler/targets.py", line 180, in <listcomp>
    return self.link([self.assemble(self.compile(module)) for module in modules],
  File "/home/britton/artiq-dev-phaser/artiq/artiq/compiler/targets.py", line 140, in compile
    llparsedmod = llvm.parse_assembly(str(llmod))
  File "/home/britton/.local/lib/python3.5/site-packages/llvmlite_artiq/binding/module.py", line 22, in parse_assembly
    raise RuntimeError("LLVM IR parsing error\n{0}".format(errmsg))
RuntimeError: LLVM IR parsing error
<string>:16332:124: error: invalid field 'subprograms'
!1160 = distinct !DICompileUnit(emissionKind: 2, runtimeVersion: 0, language: DW_LANG_Python, producer: "ARTIQ", file: !6, subprograms: !1159)
                                                                                                                           ^

whitequark, 2016-10-18:
Which llvmlite are you using?

jbqubit, 2016-10-18:
(artiq-phaser) britton@britton1:~/artiq-dev-phaser/artiq/artiq/examples/phaser$ conda list llvm
# packages in environment at /home/britton/anaconda3/envs/artiq-phaser:
#
llvm-or1k                 3.9.1                        14    m-labs/label/main
llvmlite-artiq            0.10.0.dev              py35_24    m-labs/label/main

jbqubit, 2016-10-18:
Looks like I should be using llvm-or1k =3.8.
   https://github.com/m-labs/artiq/blob/phaser/README_PHASER.rst
I downgraded. To confirm versions....

```
(artiq-phaser) britton@britton1:~/artiq-dev-phaser/artiq/artiq/examples/phaser$ conda list migen
# packages in environment at /home/britton/anaconda3/envs/artiq-phaser:
#
migen                     0.4             py_1+gitb730caf    m-labs/label/main
(artiq-phaser) britton@britton1:~/artiq-dev-phaser/artiq/artiq/examples/phaser$ conda list misoc
# packages in environment at /home/britton/anaconda3/envs/artiq-phaser:
#
misoc                     0.4                       <pip>
(artiq-phaser) britton@britton1:~/artiq-dev-phaser/artiq/artiq/examples/phaser$ conda list llvm-or1k
# packages in environment at /home/britton/anaconda3/envs/artiq-phaser:
#
llvm-or1k                 3.8.1                        18    m-labs/label/main
(artiq-phaser) britton@britton1:~/artiq-dev-phaser/artiq/artiq/examples/phaser$ conda list rust-core-or1k
# packages in environment at /home/britton/anaconda3/envs/artiq-phaser:
#
rust-core-or1k            1.12.0                        7    m-labs/label/main
(artiq-phaser) britton@britton1:~/artiq-dev-phaser/artiq/artiq/examples/phaser$ conda list cargo
# packages in environment at /home/britton/anaconda3/envs/artiq-phaser:
#
cargo                     0.11.0                        0    m-labs/label/main
(artiq-phaser) britton@britton1:~/artiq-dev-phaser/artiq/artiq/examples/phaser$ conda list binutils-or1k-linux
# packages in environment at /home/britton/anaconda3/envs/artiq-phaser:
#
binutils-or1k-linux       2.27                          5    m-labs/label/main
```
```
$ artiq_run repository/dac_setup.py
====== LLVM IR (BROKEN) DUMP ======
LLVM IR (broken) dumped as /tmp/tmpon1bvix5.ll
Traceback (most recent call last):
  File "/home/britton/anaconda3/envs/artiq-phaser/bin/artiq_run", line 11, in <module>
    load_entry_point('artiq==3.0.dev0+159.gbf942fc', 'console_scripts', 'artiq_run')()
  File "/home/britton/artiq-dev-phaser/artiq/artiq/frontend/artiq_run.py", line 221, in main
    return run(with_file=True)
  File "/home/britton/artiq-dev-phaser/artiq/artiq/frontend/artiq_run.py", line 207, in run
    raise exn
  File "/home/britton/artiq-dev-phaser/artiq/artiq/frontend/artiq_run.py", line 200, in run
    exp_inst.run()
  File "/home/britton/artiq-dev-phaser/artiq/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/core.py", line 111, in run
    self.compile(function, args, kwargs, set_result)
  File "/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/core.py", line 95, in compile
    library = target.compile_and_link([module])
  File "/home/britton/artiq-dev-phaser/artiq/artiq/compiler/targets.py", line 180, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules],
  File "/home/britton/artiq-dev-phaser/artiq/artiq/compiler/targets.py", line 180, in <listcomp>
    return self.link([self.assemble(self.compile(module)) for module in modules],
  File "/home/britton/artiq-dev-phaser/artiq/artiq/compiler/targets.py", line 140, in compile
    llparsedmod = llvm.parse_assembly(str(llmod))
  File "/home/britton/.local/lib/python3.5/site-packages/llvmlite_artiq/binding/module.py", line 22, in parse_assembly
    raise RuntimeError("LLVM IR parsing error\n{0}".format(errmsg))
RuntimeError: LLVM IR parsing error
<string>:16332:43: error: invalid field 'subprograms'
!1160 = distinct !DICompileUnit(file: !6, subprograms: !1159, emissionKind: 2, language: DW_LANG_Python, runtimeVersion: 0, producer: "ARTIQ")
                                          ^

```

jbqubit, 2016-10-18:
[tmpon1bvix5.ll.gz](https://github.com/m-labs/artiq/files/537612/tmpon1bvix5.ll.gz)

whitequark, 2016-10-19:
Yes, the problem is incorrect conda dependency that permits llvm-or1k 3.9 to be installed.

jordens, 2016-10-19:
@jbqubit please confirm and show that you are using the correct versions. I can't reproduce this.

jboulder, 2016-10-19:
I confirmed versions in my second post. Should I be doing something other
than conda list?

jordens, 2016-10-19:
You need to start being a bit more diligent about your python installations:
```
  File "/home/britton/.local/lib/python3.5/site-packages/llvmlite_artiq/binding/module.py", line 22, in parse_assembly
```
That's not a conda package at all.

jordens, 2016-10-19:
You can either install the correct conda package or use the install-from-source instructions. If you do the latter, obviously don't additionally install those as conda packages.

jordens, 2016-10-19:
Do you need us to make a phaser binary package for you?

jbqubit, 2016-10-19:
resolved

closed 2016-10-19 by jbqubit