m-labs/artiq #594: phaser: dac_setup.py hanging at jesd_stpl

jbqubit, 2016-10-19:
I've added debug prints. 
```
class DACSetup(EnvExperiment):
    def build(self):
        logging.debug("build()")
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ad9154")
        self.setattr_device("sync")

    @kernel
    def run(self):
        # TODO; remove when
        # https://github.com/m-labs/jesd204b/issues/6
        # is resolved
        for i in range(99):
            logging.debug(i)
            try:
                logging.debug(i)
                self.cfg()
                return
            except:
                pass
        self.cfg()

    @kernel
    def cfg(self):
        logging.debug("cfg core reset")
        self.core.reset()
        logging.debug("cfg enabling jesd")
        self.ad9154.jesd_enable(0)
        logging.debug("cfg jesd_prbs")
        self.ad9154.jesd_prbs(0)
        logging.debug("cfg jesd_stpl")
        self.ad9154.jesd_stpl(0)
        self.busywait_us(10000)
        logging.debug("cfg jesd_enable")
        self.ad9154.jesd_enable(1)
        logging.debug("cfg 9154 init")
        self.ad9154.init()
        logging.debug("cfg dac_setup")
        self.dac_setup()
        #print("debug 2")
        self.ad9154.jesd_enable(0)
        self.busywait_us(10000)
        self.ad9154.jesd_enable(1)
        self.monitor()
```

```
$ artiq_run startup_kernel.py --verbose --verbose
DEBUG:artiq.compiler.import_cache:hook installed
DEBUG:artiq.compiler.import_cache:added '/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/ad9516_reg.py' to cache
DEBUG:artiq.compiler.import_cache:added 'startup_kernel.py' to cache
DEBUG:artiq.compiler.import_cache:added '/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/ttl.py' to cache
DEBUG:artiq.compiler.import_cache:added '/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/ad9154.py' to cache
DEBUG:artiq.coredevice.comm_tcp:connected to host 192.168.1.71 on port 1381
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.IDENT_REQUEST: 3>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.IDENT_REQUEST: 3> length=0
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.IDENT_REPLY: 2> length=24
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.SWITCH_CLOCK: 4>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.SWITCH_CLOCK: 4> length=1
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.CLOCK_SWITCH_COMPLETED: 3> length=0
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.LOAD_LIBRARY: 5>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.LOAD_LIBRARY: 5> length=10164
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.LOAD_COMPLETED: 5> length=0
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.RUN_KERNEL: 6>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.RUN_KERNEL: 6> length=0
DEBUG:artiq.coredevice.comm_generic:running kernel
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.RPC_REQUEST: 10> length=12
DEBUG:artiq.coredevice.comm_generic:rpc service: [1]<function Core.run.<locals>.set_result at 0x7fb9d2ced158> [None] {} -> b'n\x00'
DEBUG:artiq.coredevice.comm_generic:rpc service: 1 [None] {} == None
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.RPC_REPLY: 7>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.RPC_REPLY: 7> length=6
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.KERNEL_FINISHED: 7> length=0
DEBUG:artiq.coredevice.comm_tcp:disconnected
(artiq-phaser) britton@britton1:~/artiq-dev-phaser/artiq/artiq/examples/phaser$ 
(artiq-phaser) britton@britton1:~/artiq-dev-phaser/artiq/artiq/examples/phaser$ 
(artiq-phaser) britton@britton1:~/artiq-dev-phaser/artiq/artiq/examples/phaser$ artiq_run repository/dac_setup.py  --verbose --verbose
DEBUG:artiq.compiler.import_cache:hook installed
DEBUG:artiq.compiler.import_cache:added '/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/ad9154_reg.py' to cache
DEBUG:artiq.compiler.import_cache:added 'repository/dac_setup.py' to cache
DEBUG:root:build()
DEBUG:artiq.compiler.import_cache:added '/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/ttl.py' to cache
DEBUG:artiq.compiler.import_cache:added '/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/ad9154.py' to cache
DEBUG:artiq.coredevice.comm_tcp:connected to host 192.168.1.71 on port 1381
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.IDENT_REQUEST: 3>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.IDENT_REQUEST: 3> length=0
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.IDENT_REPLY: 2> length=24
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.SWITCH_CLOCK: 4>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.SWITCH_CLOCK: 4> length=1
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.CLOCK_SWITCH_COMPLETED: 3> length=0
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.LOAD_LIBRARY: 5>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.LOAD_LIBRARY: 5> length=18788
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.LOAD_COMPLETED: 5> length=0
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.RUN_KERNEL: 6>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.RUN_KERNEL: 6> length=0
DEBUG:artiq.coredevice.comm_generic:running kernel
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.RPC_REQUEST: 10> length=16
DEBUG:artiq.coredevice.comm_generic:rpc service: [2]<function debug at 0x7fa6293b4e18> [0] {} -> b'n\x00'
DEBUG:root:0
DEBUG:artiq.coredevice.comm_generic:rpc service: 2 [0] {} == None
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.RPC_REPLY: 7>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.RPC_REPLY: 7> length=6
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.RPC_REQUEST: 10> length=16
DEBUG:artiq.coredevice.comm_generic:rpc service: [2]<function debug at 0x7fa6293b4e18> [0] {} -> b'n\x00'
DEBUG:root:0
DEBUG:artiq.coredevice.comm_generic:rpc service: 2 [0] {} == None
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.RPC_REPLY: 7>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.RPC_REPLY: 7> length=6
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.RPC_REQUEST: 10> length=31
DEBUG:artiq.coredevice.comm_generic:rpc service: [2]<function debug at 0x7fa6293b4e18> ['cfg core reset'] {} -> b'n\x00'
DEBUG:root:cfg core reset
DEBUG:artiq.coredevice.comm_generic:rpc service: 2 ['cfg core reset'] {} == None
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.RPC_REPLY: 7>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.RPC_REPLY: 7> length=6
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.RPC_REQUEST: 10> length=34
DEBUG:artiq.coredevice.comm_generic:rpc service: [2]<function debug at 0x7fa6293b4e18> ['cfg enabling jesd'] {} -> b'n\x00'
DEBUG:root:cfg enabling jesd
DEBUG:artiq.coredevice.comm_generic:rpc service: 2 ['cfg enabling jesd'] {} == None
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.RPC_REPLY: 7>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.RPC_REPLY: 7> length=6
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.RPC_REQUEST: 10> length=30
DEBUG:artiq.coredevice.comm_generic:rpc service: [2]<function debug at 0x7fa6293b4e18> ['cfg jesd_prbs'] {} -> b'n\x00'
DEBUG:root:cfg jesd_prbs
DEBUG:artiq.coredevice.comm_generic:rpc service: 2 ['cfg jesd_prbs'] {} == None
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.RPC_REPLY: 7>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.RPC_REPLY: 7> length=6
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.RPC_REQUEST: 10> length=30
DEBUG:artiq.coredevice.comm_generic:rpc service: [2]<function debug at 0x7fa6293b4e18> ['cfg jesd_stpl'] {} -> b'n\x00'
DEBUG:root:cfg jesd_stpl
DEBUG:artiq.coredevice.comm_generic:rpc service: 2 ['cfg jesd_stpl'] {} == None
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.RPC_REPLY: 7>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.RPC_REPLY: 7> length=6
```
Hangs indefinitely. No timeout.

jordens, 2016-10-19:
No. I bet it hangs in `busywait_us`.
* Are you providing a 2 GHz clock with the right power to J1?
* Have you compiled and flashed the startup kernel?
* Is the startup kernel being executed?
* Did you modify the `device_db.pyon`?

jbqubit, 2016-10-20:
>Are you providing a 2 GHz clock with the right power to J1?
Yes. 
>Have you compiled and flashed the startup kernel?
I built the .elf and loaded onto flash. 

>Is the startup kernel being executed?
After resetting the board I can do $artiq_run artiq_run startup_kernel.py without error. If I then do $ $artiq_run repository/dac_setup.py it hangs and subsequent attempts to do $artiq_run artiq_run startup_kernel.py fail. 

>Did you modify the device_db.pyon?
Yes. Modified IP address of core device.

jordens, 2016-10-20:
* Did you set startup_clock as per the instructions?
* Do the boot messages show that the startup kernel is executed without error? I.e. paste the boot log.
* Measure the frequency of sysref on JP26/JP27.

jbqubit, 2016-10-20:
>Did you set startup_clock as per the instructions?

Yes. It is set as i. 

>Do the boot messages show that the startup kernel is executed without error? I.e. paste the boot log.

See snipped I pasted below which shows --verbose output from running startup kernel manually. I've inspected the boot log and don't see errors. I'm not in front of KC705 at the moment so can't paste the log. 

>Measure the frequency of sysref on JP26/JP27.

I will do this.

jordens, 2016-10-20:
Below?

jordens, 2016-10-20:
And I need what I asked for. The boot log, not you manually running something.

jordens, 2016-10-20:
If you want to confirm my suspicion without being in front of the board, you can try to count pulses on the sysref ttl channel for a µs. That should also block.

jbqubit, 2016-10-20:
The manual run is in the snippet which opened the Issue. 
```$ artiq_run startup_kernel.py --verbose --verbose```

> And I need what I asked for. The boot log, not you manually running something.

Acknowledged. Todo... 

- [ ] Measure the frequency of sysref on JP26/JP27.
- [ ] Do the boot messages show that the startup kernel is executed without error? I.e. paste the boot log.
- [ ] try to count pulses on the sysref ttl channel for a µs. That should also block.

jordens, 2016-10-20:
And just to verify, in `device_db.pyon`, `core.arguments.external_clock == True` as shipped?

jbqubit, 2016-10-20:
Can ping KC705.
```
$ artiq_corelog --verbose --verbose
DEBUG:artiq.coredevice.comm_tcp:connected to host 192.168.1.71 on port 1381
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.IDENT_REQUEST: 3>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.IDENT_REQUEST: 3> length=0
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.IDENT_REPLY: 2> length=24
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.LOG_REQUEST: 1>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.LOG_REQUEST: 1> length=0
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.LOG_REPLY: 1> length=4096
Startup RTIO clock: internal
Startup kernel started
WARNING: startup kernel ended with exception
Startup kernel terminated
DEBUG:artiq.coredevice.comm_tcp:disconnected

```
```
$ artiq_coreconfig read idle_kernel
Key idle_kernel does not exist
```

```
$ artiq_run startup_kernel.py --verbose --verbose
DEBUG:artiq.compiler.import_cache:hook installed
DEBUG:artiq.compiler.import_cache:added '/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/ad9516_reg.py' to cache
DEBUG:artiq.compiler.import_cache:added 'startup_kernel.py' to cache
DEBUG:artiq.compiler.import_cache:added '/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/ttl.py' to cache
DEBUG:artiq.compiler.import_cache:added '/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/ad9154.py' to cache
DEBUG:artiq.coredevice.comm_tcp:connected to host 192.168.1.71 on port 1381
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.IDENT_REQUEST: 3>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.IDENT_REQUEST: 3> length=0
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.IDENT_REPLY: 2> length=24
DEBUG:artiq.coredevice.comm_generic:preparing to send message: type=<_H2DMsgType.SWITCH_CLOCK: 4>
DEBUG:artiq.coredevice.comm_generic:sending message: type=<_H2DMsgType.SWITCH_CLOCK: 4> length=1
DEBUG:artiq.coredevice.comm_generic:receiving message: type=<_D2HMsgType.CLOCK_SWITCH_FAILED: 4> length=0
DEBUG:artiq.coredevice.comm_tcp:disconnected
Traceback (most recent call last):
  File "/home/britton/anaconda3/envs/artiq-phaser/bin/artiq_run", line 11, in <module>
    load_entry_point('artiq==3.0.dev0+159.gbf942fc.dirty', 'console_scripts', 'artiq_run')()
  File "/home/britton/artiq-dev-phaser/artiq/artiq/frontend/artiq_run.py", line 221, in main
    return run(with_file=True)
  File "/home/britton/artiq-dev-phaser/artiq/artiq/frontend/artiq_run.py", line 207, in run
    raise exn
  File "/home/britton/artiq-dev-phaser/artiq/artiq/frontend/artiq_run.py", line 200, in run
    exp_inst.run()
  File "/home/britton/artiq-dev-phaser/artiq/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/core.py", line 115, in run
    self.comm.switch_clock(self.external_clock)
  File "/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/comm_generic.py", line 247, in switch_clock
    self._read_empty(_D2HMsgType.CLOCK_SWITCH_COMPLETED)
  File "/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/comm_generic.py", line 138, in _read_empty
    self._read_expect(ty)
  File "/home/britton/artiq-dev-phaser/artiq/artiq/coredevice/comm_generic.py", line 134, in _read_expect
    format(self._read_type, ty))
OSError: Incorrect reply from device: _D2HMsgType.CLOCK_SWITCH_FAILED (expected _D2HMsgType.CLOCK_SWITCH_COMPLETED)
```

>verify, in device_db.pyon, core.arguments.external_clock == True

yes

>Measure the frequency of sysref on JP26/JP27.

No signal discernable on 1 GS/s scope.

jordens, 2016-10-20:
Please just do what I asked and what's suggested in the readme. Use flterm to observe the entire boot sequence and paste the result here. Otherwise this will take a very long time.
Also, you must have changed something that you did not disclose. Before, your `$ artiq_run startup_kernel.py --verbose --verbose` was succeeding. Now it is failing.
FInally, did you set the jumpers oon XP1 and JP3 as described?

closed 2016-10-24 by jbqubit