m-labs/artiq #595: digital camera/frame grabber support (Camera Link)

jordens, 2016-10-27:
Electron multiplying CCD cameras can perform as well as PMTs for many applications. They have the advantage of many more pixels, lower price per pixel, and all the obvious offerings of being able to spatially localize (i.e. image) photons.

By implementing a Camera Link receiver, an ARTIQ DRTIO satellite could within a few 100 µs receive the frame and perform cropping, binning, and photon counting on multiple ROIs in a scalable way.

http://www.andor.com/pdfs/specifications/iXon_Ultra_888_EMCCD_Specifications.pdf

http://www.xilinx.com/support/documentation/application_notes/xapp585-lvds-source-synch-serdes-clock-multiplication.pdf

jordens, 2017-08-28:
https://github.com/m-labs/sinara/wiki/Grabber

sbourdeauducq, 2018-07-24:
Done and seems to work except:
* the FPGA receives a 2048x1024 picture instead of the expected 1024x1024 for some reason. Each pixel appears to be received twice.
* the collected ROI outputs appear reasonable, but I have not checked them against the camera picture received over USB.

@jordens Can you review the code, and also look into these two points? Since you can get physical access to the camera it may be simpler.

gkasprow, 2018-07-24:
Maybe you sample each bit twice, i.e. by using DDR buffer

jordens, 2018-07-24:
A couple things now. I'll do a detailed review later:

* I think
```diff
-static mut GRABBER_STATE: &'static mut [State] = &mut [State::Down; csr::GRABBER_LEN];
+static mut GRABBER_STATE: [State; csr::GRABBER_LEN] = [State::Down; csr::GRABBER_LEN];
```
* The `WaitResolution` state should be removed. The camera is generally not used as a video camera, it's not delivering frames all the time, we can't block on that.
* If that's 2048 rows it's probably due to the frame transfer area. If it's 2048 columns, that's weird. Are you saying the same pixel **values** are received twice?

sbourdeauducq, 2018-07-24:
> The WaitResolution state should be removed. The camera is generally not used as a video camera, it's not delivering frames all the time, we can't block on that.

It won't block, it just waits and then displays whatever values are there after a fixed time.

> If that's 2048 rows it's probably due to the frame transfer area. If it's 2048 columns, that's weird. Are you saying the same pixel values are received twice?

Yes.

jordens, 2018-08-02:
Pixel doubling presumably to meet min f requirement.

closed 2018-08-02 by jordens