m-labs/artiq #596: Can't return lists in kernel

r-srinivas, 2016-10-27:
Using artiq 2.0 on windows, I have the following experiment

```python
class List_Test(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")

    @kernel        
    def run(self):
        x = self.return_list()
        print(x)
        
    @kernel
    def return_list(self):
        return [1, 2, 3]
```

Which generates the error message,

```
ERROR:worker(23359,variable_test.py):root:While compiling <repository>\test\vari
able_test.py
<repository>\test\variable_test.py:44:16-44:25: error: cannot return an allocate
d value that does not live forever
        return [1, 2, 3]
               ^^^^^^^^^
<repository>\test\variable_test.py:42:5: note: the value of the expression is al
ive from this point...
    @kernel
    ^
<repository>\test\variable_test.py:44:25: note: ... to this point
        return [1, 2, 3]
                        ^
<repository>\test\variable_test.py:44:16-44:25: note: this expression has type l
ist(elt=numpy.int32)
        return [1, 2, 3]
               ^^^^^^^^^
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 23359
```

It would be useful to be able to return lists. For example, we may want to have one method that generates a list of all the results that we then pass to another function that processes that list.

whitequark, 2016-10-28:
@sbourdeauducq How do you expect this to work? Once we return, anything allocated on the stack is destroyed.

sbourdeauducq, 2016-10-28:
I know this, but is are there useful special cases that can be implemented anyway?

sbourdeauducq, 2016-10-28:
@r-srinivas in your example, you can allocate the list in ``run``, pass it as parameter to ``return_list``, and mutate it there.

whitequark, 2016-10-28:
@r-srinivas Or even better, do something like this:
```
@kernel
def return_list(self, k):
  list = [1, 2, 3]
  k(list)

@kernel
def run(self):
  def do_with_list(l):
    print(l)
  self.return_list(do_with_list)
```

r-srinivas, 2016-10-29:
I'm not sure if that would work for the particular application we had in mind.

```python
    @kernel
    def run_scan(self):
        
        for i in range(self.l):
            self.init_scan_point(i)
            results = self.run_scan_point(i)
            self.cleanup_scan_point(i)
            self.update_datasets(i, results)
            if self.scheduler.check_pause():
                break

    @kernel
    def run_scan_point(self, index):
        self.core.break_realtime()
        results = [0]*self.repetitions
        self.ttl_bdd.off()
        for i in range(self.repetitions):
            results[i] = self.single_rep(index)
            # delay for gate rising in detection
            delay(20*us)
        self.ttl_bdd.on()
        return results
```

The work around we've been using is to preallocate the list in prepare and have it as a class variable, i.e `self.results = [0]*self.repetitions` which we just overwrite for every scan point.

sbourdeauducq, 2016-10-29:
What I proposed would work:
```python
    @kernel
    def run_scan(self):
        for i in range(self.l):
            self.init_scan_point(i)
            results = [0]*self.repetitions
            self.run_scan_point(results, i)
            self.cleanup_scan_point(i)
            self.update_datasets(i, results)
            if self.scheduler.check_pause():
                break

    @kernel
    def run_scan_point(self, results, index):
        self.core.break_realtime()
        self.ttl_bdd.off()
        for i in range(self.repetitions):
            results[i] = self.single_rep(index)
            # delay for gate rising in detection
            delay(20*us)
        self.ttl_bdd.on()
```

r-srinivas, 2016-11-01:
That works, thanks!

closed 2016-11-01 by r-srinivas