m-labs/artiq #597: RPCs don't modify class variables on host

r-srinivas, 2016-10-27:
Using artiq 2.0 on windows, I have the following experiment

```python
class Variable_Test(EnvExperiment):

    def build(self):
        self.setattr_device("core")

    def prepare(self):
        self.x = 1

    @kernel
    def run(self):
        self.x=2
        print(self.x)
        self.host_print()

    def host_print(self):
        print(self.x)

    def analyze(self):
        print(self.x)
```

The output is,

```
INFO:worker(23360,variable_test.py):print:2
INFO:worker(23360,variable_test.py):print:1
INFO:worker(23360,variable_test.py):print:2
```

When the RPC `host_print` is called, I understand that it's looking for the value of `self.x` on the host, not from the kernel, which is why it prints 1. This only gets modified once the experiment runs, for example in the `analyze` that prints 2. If an RPC is called anyway, should any rpc that calls a class variable also modifiy its value on the host based on what it is on the kernel? If not, it seems like the user could unknowingly try to process some data thinking it's what was on the kernel at that time but is in fact its original value.

sbourdeauducq, 2016-10-28:
Synchronizing all host variables at every RPC would be slow, which is why we are not doing it.

closed 2016-10-28 by sbourdeauducq

r-srinivas, 2016-10-28:
Even if it were only one way, i.e, the value from the kernel only overwrites the value on the host?

sbourdeauducq, 2016-10-28:
There isn't just "the value". There are typically lots of values, think of what would happen with a complex experiment with 40 TTLs, 20 DDS and 12 user classes. All the values inside those classes would potentially have to be sent at every RPC (there is no good way of knowing which ones the RPC method will use).

r-srinivas, 2016-10-28:
Okay, that makes sense. Could this maybe be added to the documentation? Probably the best place to do this is here https://m-labs.hk/artiq/manual-release-2/compiler.html ?