m-labs/artiq #598: [phaser] kernels freeze

jordens, 2016-10-28:
with current phaser (b14fcd4) kernels freeze depending on where I return:
If I place a return before the line that prints AD9154_SCR_L_REG, the kernel runs fine.
https://github.com/m-labs/artiq/blob/phaser/artiq/examples/phaser/repository/test_ad9154_status.py#L107
If I place it anywhere after (of using the code in the repository as is), it does not complete loading (no `_D2HMsgType.LOAD_COMPLETED` received, i.e. nothing received).
The particular line is not the problem. It seems to be more the amount of code.

Good (return before line 107) and bad (return just after line 107) dumps are attached.

[good_bad.tar.gz](https://github.com/m-labs/artiq/files/558649/good_bad.tar.gz)

jordens, 2016-10-28:
This didn't happen before i merged master.

whitequark, 2016-10-28:
I don't think this is related to the compiler. Rather this seems like yet another lwip bug.

jordens, 2016-11-13:
Still happens in 2e482505c67b3693ba16c8db4fc1d40a0269e337

whitequark, 2016-12-05:
This is a manifestation of #637, let's discuss it there.

closed 2016-12-05 by whitequark