m-labs/artiq #601: moninj clobbers RTIO TTL channels >63

klickverbot, 2016-10-31:
In multiple places, the assumption is currently made that there are less than 65 TTL channels:
 * The fields in the moninj protocol are only 64 bits wide.
 * `runtime/moninj.c` exhibits undefined behaviour with CONFIG_RTIO_REGULAR_TTL_COUNT > 64 (left-shifting by more than the width of a 64 bit integer). The Rust runtime will probably panic or mask the bits. (?)
 * The UI displays more than 64 channels, even though they can't possibly work given the protocol.

A first step would be to at least leave these channels alone – currently, lust looking at the higher channels in a funny way in the UI brigs them into an interesting state (patch WIP).

whitequark, 2016-10-31:
> The Rust runtime will probably panic or mask the bits. (?)

Both. It panics with debug assertions on (we have them on currently as they shouldn't slow the runtime down too much), and masks with debug assertions off.

dhslichter, 2016-11-01:
We will soon be implementing hardware to allow 108 TTL channels, so it would be good to have this fixed in the near term.  Based on the number of available pins on the KC705, going to a 256-bit-wide moninj seems like it should be more than sufficient.  

Aside from the moninj system, though, it seems one should be able to add RTIO channels without bound (other than hardware constraints), no?

klickverbot, 2016-11-04:
It took me longer than expected to find a few quiet minutes, but I've put up a PR that at least avoids corrupting the state from the GUI and when sending updates.

dhslichter, 2016-11-04:
@sbourdeauducq can we move to make the moninj capable of more lines, e.g. 128 or 256?  If this is coupled with #603, it should be possible to mitigate GUI issues from having additional TTL lines.

closed 2016-11-05 by sbourdeauducq