m-labs/artiq #609: kernel_invariants does not enable constant propagation as advertised

klickverbot, 2016-11-08:
Currently, `kernel_invariants` only adds `!invariant.load` to LLVM `loads` from the respective attribute. This, however, doesn't enable constant propagation of whatever their values are into the respective loads, only merging/hoisting of multiple loads from the same attribute. Even the example describing the feature from `docs/compiler.rst` isn't actually optimised:

```py
class Converter:
    kernel_invariants = {"ratio"}

    def __init__(self, ratio=1.0):
        self.ratio = ratio

    @kernel
    def convert(self, value):
        return value * self.ratio ** 2
```

(The IR for the kernel contains a load from ratio and two fmuls.)

LLVM doesn't optimise out the loads because `self` is emitted as a mutable global. Since the symbol is `private`, it could conceivably be a bit smarter and detect that there are no writes in the module and no mutable aliases escaped, allowing it to still do the optimisation.

I haven't looked into the LLVM side of things more closely yet, but either way, a lot of optimisation potential is currently wasted (LLVM 3.8). This is a serious performance issue in latency-sensitive experiments of even just moderate complexity – at least unless one wants to manually inline everything.

As a quick fix for an experiment we are running, I've hacked up a PoC patch that emits objects consisting solely of `kernel_invariants` as actual constants (PR incoming). This is not quite as flexible as one would hope, though. Proper constant propagation would be preferable, but I wasn't quite sure where to best fit it into the design.

klickverbot, 2016-11-08:
(See #610 for the PoC pull request.)

whitequark, 2016-11-09:
Now the example in the documentation that you've called out is correct, and I've added another one, which illustrates how kernel_invariants helps LICM.

commit eee8d053 (2016-11-09): doc: clarify kernel_invariant doc (fixes #609).
closed 2016-11-09 by whitequark