m-labs/artiq #611: phaser: with parallel set_phase() not aligned

jbqubit, 2016-11-09:
using phaser branch, commit b14fcd4   

Following produces phase-locked sinusoids with a phase offset of 2pi*0.2 rad. Expectation is phase offset of 0 rad. 

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.break_realtime()
        self.ttl_sma.output()
        delay(1*ms)

        v = 1.0
        f = 100*MHz

        self.sawg0.set_amplitude(v)
        self.sawg0.set_frequency(f)
        self.sawg1.set_amplitude(v)
        self.sawg1.set_frequency(f)

        with parallel:
            self.sawg0.set_phase(0.0)
            self.sawg0.set_phase(0.0)
```

jordens, 2016-11-09:
I don't understand your logic behind the reasoning of what you expect.
All `set_*()` ops are zero-duration.
Please check your code and verify that you want to set the same phase twice on the same channel at the same phase.

jordens, 2016-11-09:
Oh and please stick with the earlier know working commits, i.e. https://github.com/m-labs/artiq/tree/c428800caf52a77a22a9be0d967715cccb4e0b63

jordens, 2016-11-13:
@jbqubit ping

jordens, 2017-01-26:
no response

closed 2017-01-26 by jordens