m-labs/artiq #612: phaser: AD9154-FMC not phase locked to KC705

jbqubit, 2016-11-09:
using phaser branch, commit b14fcd4

Expectation is that pulses produced by KC705 are phase locked to RF generated by AD9154-FMC. Observation is no well-defined phase relationship. 

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.break_realtime()
        self.ttl_sma.output()
        delay(1*ms)

        v = 1.0
        f = 100*MHz

        self.sawg0.set_amplitude(v)
        self.sawg0.set_frequency(f)
        self.sawg1.set_amplitude(v)
        self.sawg1.set_frequency(f)

        while True:
            with parallel:
                self.sawg0.set_phase(0.0)
                self.sawg0.set_phase(0.0)
                self.ttl_sma.pulse(1*us)
            delay(9*us)
```

jordens, 2016-11-09:
It's unclear what is not "phase locked" to what. I.e. do you mean a delay variation or an actual drift when running.
The delay variation is known and expected. It requires active realignment of sysref.

closed 2016-11-09 by jordens

jbqubit, 2016-11-09:
> It's unclear what is not "phase locked" to what. I.e. do you mean a delay non-determinism or an actual drift when running?

If I trigger my scope on ttl_sma I see that the phase of the sawgx is not deterministic. 

> The delay non-determinism is known and expected. 

In this case it's not possible to do the "absolute phase" test I had planned https://github.com/m-labs/sinara/issues/28. @jmizrahi do you need TTL-pulse to DAC determinism for your synchronization demo (pair of KC705 and AD9154-FMC)?

jordens, 2016-11-09:
No you can't time DAC outputs to TTLs deterministically. That's expected. As I said before, this will come in phaser2.

jbqubit, 2016-11-09:
"As I said before, this will come in phaser2." lol I don't know when this was said. But that's OK. I'll do this test when phaser2 is available.

jordens, 2016-11-09:
To do multi chip synchronization with independent clock trees or synchronization to other RTIO channels we need to actively align the sysrefs. We have described the strategy to do MCS with independent boards and DRTIO in Warsaw and on the mailing list.