m-labs/artiq #616: Pipistrello on windows?

edowd, 2016-11-15:
Hi - I am just getting started with Artiq and a pipistrello board on a windows computer. I have been following the directions and finished the step of flashing the board. The next step in the manual is "configuring the core device," but my understanding is this is for the KC705, not the pipistrello - is that correct? For the pipistrello I am trying to run a PPP session over the serial port as described here (https://m-labs.hk/artiq/manual-release-2/core_device.html#) but it says here Windows is not supported - does that mean you can't use Artiq with Pipistrello on Windows? What am I missing?

closed 2016-11-16 by edowd

sbourdeauducq, 2016-11-16:
You can either figure out how to use the Windows PPP API to skip the ISP/modem stuff (if possible at all), or write code on the Pipistrello to make Windows think it is talking to an ISP.

edowd, 2016-11-16:
Thanks - I'm using a virtual linux machine instead.