m-labs/artiq #618: drtio: implement soft-CPU in satellite design

sbourdeauducq, 2016-11-16:
It will be useful to have a CPU in the Sayma DRTIO satellite to perform the following tasks:
* Si5324 programming (as currently done by the gateware sequencer)
* DDR3 SDRAM bring-up (initialization sequence, read/write leveling as implemented in the MiSoC BIOS)
* answering auxiliary DRTIO packets: identification of gateware/hardware module, moninj, (later) reflashing, (later) readout of on-board temperature sensors

Since none of this code is performance-critical, and it should use little RAM, it can be simpler to have a single-stage firmware like the MiSoC BIOS: executed-in-place for code and using block RAM for data. If using Rust, check its RAM footprint as block RAM is expensive and I'd rather keep it below 32KB. Such a single-stage firmware is compatible with any future devices that do not have SDRAM, e.g. potentially Kasli.

The MiSoC build system should be reworked to allow alternative "BIOSes" and sharing of the SDRAM initialization code.

whitequark, 2016-11-16:
> If using Rust, check its RAM footprint as block RAM is expensive and I'd rather keep it below 32KB

Statically linked "base Rust" (i.e. using only libcore) has zero inherent RAM footprint, and you only pay for what you use, in terms of data structures (which you define).

jordens, 2016-11-16:
Also:

  * delay scan and measurement for the active phase alignment, storage of the fractional part (could also be upstream)

IMHO we could also consider going the exact opposite way and try not to add too many special case distinctions between DRTIO master and slave. Is this strict distinction really needed?

sbourdeauducq, 2016-11-16:
> Is this strict distinction really needed?

Yes. A number of things are already shared though.

jordens, 2016-11-16:
The next question would then be: Why do you think strict distinction is really needed?

sbourdeauducq, 2016-11-16:
Many reasons; support devices without SDRAM, no need for two CPUs, different constraints, etc.

jordens, 2016-11-16:
My prediction would be that the satellites including Kasli will all have SDRAM, that there will soon be a need to perform the typical comms cpu functionality (save to flash, start/stop remote kernels, manage datasets etc), that there will soon be a CPU for remote kernels, etc. And then this distinction would be shortsighted. "Different constraints" doesn't mean much to me.

sbourdeauducq, 2016-11-16:
If there are remote kernels then DRTIO is not the right protocol.

jordens, 2016-11-16:
In a device where RTIO is driven by multiple different masters (upstream DRTIO master, multiple DMA engines, kernel API) I can see DRTIO coexisting happily with remote kernels. Especially since the arbitration between DMA and kernel API needs to be done anyway.

sbourdeauducq, 2016-11-16:
There are no plans for such things right now, and those remote kernels will most likely want to pass messages between them which is nontrivial (there's all the compiler/software support, plus e.g. they may have to cross switches which can cause packet collisions). Can we please just get a simple system to work right now?

sbourdeauducq, 2016-11-16:
If all the satellites have SDRAM, the current boot mechanism is fine though.

closed 2017-02-22 by sbourdeauducq