m-labs/artiq #619: DRTIO switch support

sbourdeauducq, 2016-11-16:


sbourdeauducq, 2017-01-30:
For supporting the Sinara hardware, switches should be capable of data rate conversion, as the AMC-RTM link has lower bandwidth than the backplane links.

jbqubit, 2017-02-09:
@sbourdeauducq Do you have enough information to propose an implementation and quote this feature?

sbourdeauducq, 2017-05-09:
Funded by ARL for the purposes of supporting the RTM FPGA.

hartytp, 2018-12-26:
@sbourdeauducq am I right in thinking that this is no longer needed, since we will use transceivers for RTM DRTIO on v2.0 hw? In which case, does that free up some funding for other purposes?

sbourdeauducq, 2018-12-27:
@hartytp DRTIO switching is partly funded through your contract and through the Sayma contract (this Issue is about switching support, not transceiver-related issues).
Indeed, we don't need rate conversion support anymore; though that's not particularly hard to do upper in the stack with the current DRTIO switching architecture, the Ultrascale IOSERDES turned out to be a major source of problems (TPWS+BUFGCE_DIV).

hartytp, 2018-12-27:
ack

So, if we scrap rate conversion then this item is complete and the issue can be closed?

sbourdeauducq, 2018-12-27:
Yes, it'll be regular DRTIO switching on the RTM.

closed 2018-12-27 by sbourdeauducq