m-labs/artiq #622: seconds_to_mu should take the same in prepare and on the core device

dleibrandt, 2016-11-17:
The following code throws an error, because `seconds_to_mu` requires different arguments in prepare and on the core device.  It's annoying to write two versions of `set_pulse_time` so that I can call it from both locations.  Would it be possible to make `seconds_to_mu` on the core device accept the `core` argument, even if it doesn't use it?

```
class UnitConversionTime1(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        
    def prepare(self):
        self.set_pulse_time(1.*us)
        
    @portable
    def set_pulse_time(self, t):
        self.t = t
        self.t_mu = seconds_to_mu(self.t, core=self.core)

    @kernel
    def run(self):
        self.set_pulse_time(1.*us)
```

sbourdeauducq, 2016-11-18:
Alternatively ``seconds_to_mu`` and ``mu_to_seconds`` could be made methods of the core device object.

@dleibrandt If you notice any performance improvement with this technique, please post your code as I suspect that this is something that the compiler should have done itself e.g. with inlining/constant propagation.

whitequark, 2016-11-18:
> Alternatively seconds_to_mu and mu_to_seconds could be made methods of the core device object.

This seems preferable.

dleibrandt, 2016-11-18:
@sbourdeauducq 's suggestion sounds good to me.

I do see a significant speedup.  A simplified example is below:
```
class UnitConversionTime3(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl12")
        
    def prepare(self):
        self.set_time(1000.*us)
        
    def set_time(self, t):
        self.t = t
        self.t_mu = seconds_to_mu(self.t, core=self.core)

    @kernel
    def set_time_kernel(self, t):
        self.t = t
        self.t_mu = seconds_to_mu(self.t)

    @kernel
    def run(self):
        for t in [20*us, 15*us, 10*us, 5*us, 2*us, 1*us]:
            print("t =", t, " us")
            self.set_time_kernel(t)
            
            self.core.reset()
            
            for _ in range(10000):
                #self.ttl12.pulse(self.t)
                self.ttl12.pulse_mu(self.t_mu)
```

With the `pulse` method commented and the `pulse_mu` method uncommented, RTIOUnderflow occurs when the pulse time is 1 us.  With the `pulse` method uncommented and the `pulse_mu` method commented, RTIOUnderflow occurs when the pulse time is 20 us.

jordens, 2016-11-18:
maybe also affected by #591

whitequark, 2016-11-20:
@dleibrandt The primary problem in your case is that you haven't marked `t` as kernel invariant. Once this is done, the `pulse` and `pulse_mu` variants generate exactly identical code.

If you are pre-calculating values such as timings, it is best to pass them through arguments. Storing them in attributes inside the kernel will hinder the optimizer.

Do you specifically want to store such calculated values in attributes? Can I see a more realistic example so that I can try to make it more ergonomic, within the constraints of the optimizer?

whitequark, 2016-11-20:
@dleibrandt To help debug these issues, commit f5cca6b includes a pass that automatically detects attributes that may be marked kernel invariant:
```
conv3.py:30:33-30:39: note: attribute 't' of type 'artiq_run_conv3.UnitConversionTime3' is never written to; it could be marked as kernel invariant to potentially increase performance
                self.ttl1.pulse(self.t)
                                ^^^^^^
```

sbourdeauducq, 2016-11-20:
Kernel invariants won't actually help here, ``t`` is not one when ``set_time_kernel`` is used.

whitequark, 2016-11-21:
Titular issue fixed in 009d396; I am addressing the underlying reason for slowness in https://github.com/m-labs/artiq/issues/624.

closed 2016-11-21 by whitequark