m-labs/artiq #623: Increase RTIO FIFO depths in class NIST_CLOCK

dleibrandt, 2016-11-18:
I've had to increase the RTIO FIFO depths in the NIST_CLOCK gateware in order to get my sideband cooling pulse sequence to run without underflow errors.  Could you please incorporate the changes in the diff below so that I don't have to continue compiling my own gateware in the future?  (I'm not aware of anyone else using the NIST_CLOCK gateware, so I don't think this would adversely affect anyone.  If I'm wrong, please comment here.)

[nist_clock_fifo_patch.txt](https://github.com/m-labs/artiq/files/600545/nist_clock_fifo_patch.txt)

For reference, even with my increased FIFO depths the block ram usage is only 47%.

As an aside, I know that @r-srinivas is doing the same thing for NIST_QC2 (see #530), so it might be convenient to him to change those FIFO depths also.

sbourdeauducq, 2016-11-20:
What is that sideband cooling pulse sequence that causes trouble?

sbourdeauducq, 2016-11-20:
@dleibrandt your patch fails timing.

sbourdeauducq, 2016-11-20:
...on master, but passes on release-2.

sbourdeauducq, 2016-11-20:
So this will stay in release-2 and be available with 2.1 unless we have problems with it again, and will not be in 3.x.

closed 2016-11-20 by sbourdeauducq

r-srinivas, 2016-11-21:
@sbourdeauducq any idea why the timing fails?

dleibrandt, 2016-11-21:
@sbourdeauducq, the sideband cooling pulse sequence is up to 1000 pulses.  The pulse durations change during the sequence, but the majority are roughly 1 us.  Each pulse consists of turning a ttl on and off and setting a dds frequency and amplitude twice (once at the start and once at the end of the pulse; we do this so that we can keep the AOM on but detune the light from the atomic resonance, thus keeping the temperature of the AOM constant).

I suspect that my underflow problem will be solved once DMA is finished.  Will DMA be included in 3.x, or will it come later?

sbourdeauducq, 2016-11-22:
DMA should be in 3.0.

sbourdeauducq, 2016-11-22:
Support for AD9914 DDS phase modes in DMA sequences is currently unclear though.

r-srinivas, 2016-11-30:
@sbourdeauducq I was thinking of submitting a patch for the nist_qc2 gateware as well. Why does the timing fail in 3.0 exactly?

sbourdeauducq, 2016-12-01:
> Why does the timing fail in 3.0 exactly?

Failing timing is a regular, semi-random and annoying problem with FPGA tools. I didn't keep the Vivado timing report, you can run the synthesis again to get it and see where the critical paths are in the design.

commit 99104499 (2016-11-20): gateware: increase RTIO FIFO sizes for NIST_CLOCK. Closes #623
commit 4a62e09b (2016-11-20): gateware: increase RTIO FIFO sizes for NIST_CLOCK. Closes #623
commit ad1049d5 (2016-11-20): Revert "gateware: increase RTIO FIFO sizes for NIST_CLOCK. Closes #623"

This reverts commit 4a62e09bd4f6a1a692d770990dbd93ac80111057.