m-labs/artiq #624: Use TBAA metadata to tell LLVM about the (lack of) aliasing between attributes

whitequark, 2016-11-20:
I'm not sure that this will work, but if it does, this can potentially lead to significant benefits because e.g. TTLOut.channel could be hoisted out of an inner loop without even declaring it as kernel invariant (since it will be no-alias wrt anything written in the loop).

whitequark, 2016-11-21:
Test case: this should result in read of `t` being hoisted out of the inner loop:
```
class UnitConversionTime3(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl1")

    @kernel
    def set_time_kernel(self, t):
        self.t = t
        self.t_mu = seconds_to_mu(self.t)

    @kernel
    def run(self):
        for t in [20*us, 15*us, 10*us, 5*us, 2*us, 1*us]:
            print("t =", t, " us")
            self.set_time_kernel(t)

            self.core.reset()

            for _ in range(10000):
                self.ttl12.pulse(self.t)
```

sbourdeauducq, 2021-12-13:
Transferred to NAC3

closed 2021-12-13 by sbourdeauducq