m-labs/artiq #626: Check if ttl channel overridden from experiment code

r-srinivas, 2016-11-23:
Is there anyway of checking if a ttl channel has been overridden from experimental code? Sometimes it's easy to forget that one has overridden some channel to debug when running an experiment. It would be nice if there was some way of checking that and flagging it/raising an exception if a relevant ttl channel has been overridden.

jordens, 2016-11-23:
I strongly advise against implementing that functionality in the way you describe it. It would violate the clear separation of functionality. Since stuff can become overridden at any time you would never be able to reliably detect that situation.

What about these alternatives:

  1. Use the artiq API to script moninj. Should even work fine from experiments.
  2. Figure out why/whether the moninj GUI does tell you clearly enough what's overridden.

r-srinivas, 2016-11-28:
To be more explicit, I guess if we can just check it at the start of the experiment at prepare that would be more than enough. Does scripting moninj already work?

The issue I had with the gui is that even with just 40 channels as we have now we can't see all of them at once so it's easy to miss channels that have been overridden.

jordens, 2016-11-28:
But for that issue it would seem you want something entirely different: a more compact moninj UI.

r-srinivas, 2016-11-28:
Making the gui more compact doesn't seem scalable especially if we upgrade the our backplane on the HPC side to have more ttl channels. I think the solution should be independent of the gui. If there's some way to check if a specific ttl channel were overridden that would help. It's not a high priority but it helps you avoid silly errors.

The mistake I ran into was that we were debugging ttl transport at some point and I had overridden the pdq_trigger high just to check if all the slaves were still responding. I forgot that was overridden when I tried transport and lost the ion as now the transport was of course not being triggered. It would have been nice if the experiment had some way of throwing an exception before the transport experiment started.