m-labs/artiq #627: Thorlabs Tdc error

danielkienzlerNIST, 2016-11-24:
running the controller:
(artiq-main_20161103_qc2) C:\Users\Rabi\Desktop>thorlabs_tcube_controller -P tdc
001 -d COM5 --bind 132.163.182.102 -p 3270

and then using artiq_rpctool to control the controller:
(artiq-main_20161103_qc2) C:\Users\Rabi>artiq_rpctool 132.163.182.102 3270 call get_status_bits
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\Scripts\artiq_rpctool-script.py", line 10, in <module>
    load_entry_point('artiq==2.0', 'console_scripts', 'artiq_rpctool')()
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\lib\site-packages\artiq\frontend\artiq_rpctool.py", line 139, in main
    call_method(remote, args.method, args.args)
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\lib\site-packages\artiq\frontend\artiq_rpctool.py", line 83, in call_method
    ret = method(*[eval(arg) for arg in args])
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\lib\site-packages\artiq\protocols\pc_rpc.py", line 177, in proxy
    return self.__do_rpc(name, args, kwargs)
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\lib\site-packages\artiq\protocols\pc_rpc.py", line 169, in __do_rpc
    return self.__do_action(obj)
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\lib\site-packages\artiq\protocols\pc_rpc.py", line 163, in __do_action
    raise_packed_exc(obj["exception"])
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\lib\site-packages\artiq\protocols\packed_exceptions.py", line 42, in raise_packed_exc
    raise exc
AttributeError: '_WindowsSelectorEventLoop' object has no attribute '_proactor'

Tried several Thorlabs diver versions including the one used by Raghu.
Artiq version: 2.0
Works with artiq 1.0

sbourdeauducq, 2016-11-24:
Can you try applying the patch above?

sbourdeauducq, 2016-11-24:
And FYI ARTIQ does not use the Thorlabs driver.

r-srinivas, 2016-11-25:
> And FYI ARTIQ does not use the Thorlabs driver.

On Windows, can you change the thorlabs device to use a virtual com port without installing the driver?

danielkienzlerNIST, 2016-11-25:
(artiq-main_20161103_qc2) C:\Users\Rabi>artiq_rpctool 132.163.182.102 3270 call
get_status_bits
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\Scripts\artiq_rpctool-script.p
y", line 10, in <module>
    load_entry_point('artiq==2.0', 'console_scripts', 'artiq_rpctool')()
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\lib\site-packages\artiq\fronte
nd\artiq_rpctool.py", line 139, in main
    call_method(remote, args.method, args.args)
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\lib\site-packages\artiq\fronte
nd\artiq_rpctool.py", line 83, in call_method
    ret = method(*[eval(arg) for arg in args])
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\lib\site-packages\artiq\protoc
ols\pc_rpc.py", line 178, in proxy
    return self.__do_rpc(name, args, kwargs)
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\lib\site-packages\artiq\protoc
ols\pc_rpc.py", line 170, in __do_rpc
    return self.__do_action(obj)
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\lib\site-packages\artiq\protoc
ols\pc_rpc.py", line 164, in __do_action
    raise_packed_exc(obj["exception"])
  File "C:\Anaconda3\envs\artiq-main_20161103_qc2\lib\site-packages\artiq\protoc
ols\packed_exceptions.py", line 42, in raise_packed_exc
    raise exc
AttributeError: '_WindowsSelectorEventLoop' object has no attribute '_proactor'

sbourdeauducq, 2016-11-26:
Are you sure the patch is applied correctly to the ARTIQ installation you are running?

sbourdeauducq, 2016-11-26:
You need to edit ``C:\Anaconda3\envs\artiq-main_20161103_qc2\lib\site-packages\artiq\protoc
ols\pc_rpc.py``

danielkienzlerNIST, 2016-11-29:
I'm not sure, but I wouldn't know if not.
Yes, I edited that file. Is there something I'm missing?
The console printout also shows different line numbers, so I guess it is using the modified file

sbourdeauducq, 2016-11-30:
> I'm not sure, but I wouldn't know if not.

Add a ``print`` into that file to check that ``ProactorEventLoop`` is correctly created and used.

sbourdeauducq, 2016-12-08:
@danielkienzlerNIST ping

danielkienzlerNIST, 2016-12-08:
code:
    See ``Server`` for a description of the parameters.
    """
    if os.name == "nt":
        loop = asyncio.ProactorEventLoop()
        asyncio.set_event_loop(loop)
        print("simple_server_loop:1")
    else:
        loop = asyncio.get_event_loop()
        print("simple_server_loop:2")
    try:

out:
simple_server_loop:1

sbourdeauducq, 2016-12-12:
You could have just done print(loop).

And you still see the problem after it has printed that?

sbourdeauducq, 2016-12-12:
Fixed.

closed 2016-12-12 by sbourdeauducq