m-labs/artiq #629: Selecting new applet on empty applet dock in artiq 3.0 crashes dashboard

r-srinivas, 2016-11-28:
Using 3.0 on windows, if you right click on the applet dock and click new group it crashes the dashboard. The error message I get is,

```
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\site-packages\artiq\gui\applets.p
y", line 542, in new_with_parent
    cb(parent=parent, **kwargs)
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\site-packages\artiq\gui\applets.p
y", line 523, in new_group
    QtCore.Qt.ItemIsDropEnabled |
AttributeError: type object 'Qt' has no attribute 'ItemIsAutoTristate'
```

r-srinivas, 2016-11-28:
Probably related, following the example here (https://github.com/m-labs/artiq/blob/master/artiq/examples/master/repository/code_applet.py), I run into the following error,

```
ERROR:dashboard:artiq.dashboard.applets_ccb:failed to process CCB
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\site-packages\artiq\dashboard\app
lets_ccb.py", line 246, in ccb_notify
    self.ccb_create_applet(*args, **kwargs)
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\site-packages\artiq\dashboard\app
lets_ccb.py", line 178, in ccb_create_applet
    parent, applet = self.locate_applet(name, group, True)
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\site-packages\artiq\dashboard\app
lets_ccb.py", line 131, in locate_applet
    new_parent = self.new_group(g, parent=parent)
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\site-packages\artiq\gui\applets.p
y", line 523, in new_group
    QtCore.Qt.ItemIsDropEnabled |
AttributeError: type object 'Qt' has no attribute 'ItemIsAutoTristate'
```

If I modify the code to `group=None` it works.

sbourdeauducq, 2016-11-29:
https://git.merproject.org/mer-core/qtbase/commit/ae8406d82f541f6d9112bdac192e5e4e114d56aa

commit c5b55c1d (2016-11-29): applets: compatibility with older Qt. Closes #629
commit d5d17aca (2016-11-29): Merge remote-tracking branch 'm-labs/master' into phaser2

* m-labs/master:
  applets: compatibility with older Qt. Closes #629
  doc: update LLVM configure command (fixes #628).
  runtime: match argument signedness between ARTIQ Python and ksupport.
  runtime: refactor rtio_output_list.
  runtime: use correct ABI when accepting ARTIQ lists.
closed 2016-11-29 by sbourdeauducq