m-labs/artiq #630: ERROR:dashboard:quamash.QEventLoop:Task exception was never retrieved when disabling applet on dashboard

r-srinivas, 2016-11-28:
Using artiq 3.0 on windows, when disabling an applet on the gui, it closed but generated this error in the log,

```
ERROR:dashboard:quamash.QEventLoop:Task exception was never retrieved
future: <Task finished coro=<AsyncioParentComm._autoclose() done, defined at C:\
Anaconda3\envs\artiq-2016-11-25\lib\site-packages\artiq\protocols\pipe_ipc.py:11
8> exception=OSError(9, 'The handle is invalid', None, 6, None)>
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\asyncio\tasks.py", line 239, in _
step
    result = coro.send(None)
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\site-packages\artiq\protocols\pip
e_ipc.py", line 120, in _autoclose
    self.server[0].close()
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\asyncio\windows_events.py", line
283, in close
    pipe.close()
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\asyncio\windows_utils.py", line 1
57, in close
    CloseHandle(self._handle)
OSError: [WinError 6] The handle is invalid
```

This happened with the big_number app and also the custom_applet.py example. Enabling and disabling the app generates this error most of the time, and can even lead to the dashboard crashing with this error message,

```
WARNING:dashboard:quamash.QEventLoop:Event callback failed: [WinError 995] The I
/O operation has been aborted because of either a thread exit or an application
request
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\site-packages\Quamash-0.5.5-py3.5
.egg\quamash\_windows.py", line 41, in _process_events
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\asyncio\windows_events.py", line
437, in finish_recv
    return ov.getresult()
OSError: [WinError 995] The I/O operation has been aborted because of either a t
hread exit or an application request

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\site-packages\Quamash-0.5.5-py3.5
.egg\quamash\_windows.py", line 44, in _process_events
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\asyncio\windows_events.py", line
79, in set_exception
    super().set_exception(exception)
  File "C:\Anaconda3\envs\artiq-2016-11-25\lib\asyncio\futures.py", line 341, in
 set_exception
    raise InvalidStateError('{}: {!r}'.format(self._state, self))
asyncio.futures.InvalidStateError: CANCELLED: <_OverlappedFuture cancelled>
```

In addition, it seems the `disable_applet` command from the `ccb` doesn't seem to be working. Running the example code ,https://github.com/m-labs/artiq/blob/master/artiq/examples/master/repository/code_applet.py, should close the applet at the end and uncheck the box on the dashboard right? Neither of those things seem to happen.

sbourdeauducq, 2016-11-29:
``disable_applet`` does work correctly here. Did you modify the ARTIQ code at all to try to fix #629? Have you tried on Linux?

sbourdeauducq, 2016-11-29:
And is the first problem also on 2.0?

r-srinivas, 2016-11-29:
I didn't modify the code to try and fix the other issue. I'll try and fix it on linux. This problem has been on 2.0 but I didn't look into it in detail because I thought it was maybe from the custom applets we have, and aside from producing an error message on the log doesn't really do anything terrible.

r-srinivas, 2016-11-29:
Happens on 2.0 as well

```
ERROR:dashboard:quamash.QEventLoop:Task exception was never retrieved
future: <Task finished coro=<AsyncioParentComm._autoclose() done, defined at C:\
Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\protocols\pipe_ipc.py:11
8> exception=OSError(9, 'The handle is invalid', None, 6, None)>
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\asyncio\tasks.py", line 239, in _
step
    result = coro.send(None)
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\protocols\pip
e_ipc.py", line 120, in _autoclose
    self.server[0].close()
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\asyncio\windows_events.py", line
283, in close
    pipe.close()
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\asyncio\windows_utils.py", line 1
57, in close
    CloseHandle(self._handle)
OSError: [WinError 6] The handle is invalid
```

Enabling and disabling big number applet.

r-srinivas, 2016-11-29:
>disable_applet does work correctly here. Did you modify the ARTIQ code at all to try to fix #629? Have you tried on Linux?

I tried on the virtual machine as well and it doesn't seem to disable either. The applet runs and I can see the countdown from 9 to 0 but then it just stays at 0 and the applet is still enabled. I'm using the exact some code as you have in the examples as far as I can tell.

jordens, 2017-05-11:
@r-srinivas @sbourdeauducq re-visit (for new Qt)?

sbourdeauducq, 2021-12-13:
Does this still occur?