m-labs/artiq #634: pause experiment bug

dleibrandt, 2016-11-30:
I'm trying to interrupt a low priority experiment with a high priority experiment.

I'm running 2.0 on linux, and attempting to run the following experiments:
```
from time import sleep
from artiq.experiment import *

class ExpA(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("scheduler")
        
    def run(self):
        while True:
            self.run_kernel()
            
            self.scheduler.pause()
    
    @kernel
    def run_kernel(self):
        while True:
            print("ExpA run")
            if self.scheduler.check_pause():
                break
            sleep(1)
            
class ExpB(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        
    @kernel
    def run(self):
        print("ExpB run")
```

First, I submit ExpA with priority 0.  Second, I submit ExpB with priority 1.  I get the following error:
```
INFO:worker(3491,TestHandover.py):print:ExpA run
INFO:worker(3491,TestHandover.py):print:ExpA run
INFO:worker(3491,TestHandover.py):print:ExpA run
INFO:worker(3491,TestHandover.py):print:ExpA run
INFO:worker(3491,TestHandover.py):print:ExpA run
INFO:worker(3492,TestHandover.py):print:ExpB run
ERROR:worker(3491,TestHandover.py):root:Terminating with exception (ConnectionResetError: Connection closed)
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/artiq-2.0/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 228, in main
    exp_inst.run()
  File "/home/rabi/dev/artiq_clocks/Testing/TestHandover.py", line 11, in run
    self.run_kernel()
  File "/home/rabi/anaconda3/envs/artiq-2.0/lib/python3.5/site-packages/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rabi/anaconda3/envs/artiq-2.0/lib/python3.5/site-packages/artiq/coredevice/core.py", line 118, in run
    self.comm.load(kernel_library)
  File "/home/rabi/anaconda3/envs/artiq-2.0/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 299, in load
    self._read_empty(_D2HMsgType.LOAD_COMPLETED)
  File "/home/rabi/anaconda3/envs/artiq-2.0/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 137, in _read_empty
    self._read_header()
  File "/home/rabi/anaconda3/envs/artiq-2.0/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 109, in _read_header
    (sync_byte, ) = struct.unpack("B", self.read(1))
  File "/home/rabi/anaconda3/envs/artiq-2.0/lib/python3.5/site-packages/artiq/coredevice/comm_tcp.py", line 60, in read
    raise ConnectionResetError("Connection closed")
ConnectionResetError: Connection closed
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 3491
```

klickverbot, 2016-11-30:
If I'm not mistaken, you currently need to close the core device connection manually by design (`self.core.comm.close()`) before calling `Scheduler.pause()`.

dleibrandt, 2016-11-30:
Yup, that does it.  Thanks.

closed 2016-11-30 by dleibrandt