m-labs/artiq #635: Deselect group in applet dock

r-srinivas, 2016-12-01:
Using 3.0 on windows, if you only have one group in the applet dock and click on it, is there any way of deselecting it? If not, trying to create applets or groups from the GUI it automatically creates it within that group and there's no way of going one level higher.

Explicitly, from an empty applet dock,

1. Right click new group
2. Click on that group
3. Right click new applet/new group

sbourdeauducq, 2016-12-01:
You can drag the applet out of the group to the top-level.

sbourdeauducq, 2016-12-01:
And as often, better behavior cannot be obtained without wrangling with Qt:
http://doc.qt.io/qt-5/qabstractitemview.html#SelectionMode-enum
"When the user selects an item, any already-selected item becomes unselected, and **the user cannot unselect the selected item by clicking on it**."

r-srinivas, 2016-12-01:
Okay. If we can just drag it out that's easy enough.

sbourdeauducq, 2016-12-01:
Let's keep this issue open - small UI improvements like this, even though they are a PITA to program, can save some frustration to the user.

SimonRenblad, 2024-08-01:
This is a non-issue in PyQt6. Simply Ctrl+Left Click and it will deselect the group. https://doc.qt.io/qt-6/qabstractitemview.html#SelectionMode-enum. See notes on `SingleSelection`. I have tested it in `artiq_dashboard` without problems.

architeuthidae, 2024-08-01:
Re documentation, thoughts: 

(1) This can certainly be mentioned in a future expansion of the management tutorial (something I think isn't a bad idea, since the current manual lets a lot of pretty sizable features go totally unmentioned). However, that's a single keyboard-shortcut-esque detail buried in what will be a pretty long document, and I doubt that people would retain it on a first reading or necessarily remember to check the manual afterwards. Some similar details are currently in the FAQ, which I don't know if that's an intuitive place to check either...

(2) Thinking about what _is_ a natural place to look at for this kind of thing as a user of a GUI, and I think my best two ideas are (a) add 'Deselect' to the right-click menu with a 'Ctrl+click' shortcut tip, though I don't know enough about Qt6 to guess if that'd be painful, or (b) add some semitransparent help text to a corner of the background of the Applet list field itself which says, e.g. 'Click to select, Ctrl+click to deselect'. It does apply to both applets and groups, at least.

SimonRenblad, 2024-08-01:
I also considered 2a as a possible solution, however since it is not strictly a keyboard shortcut (QShortcut) but a mouse event with a modifier (QMouseEvent), it may lead to some inconsistency. Additionally, it invites the question of whether ctrl+scrolling should be documented in the case of a file explorer e.g in `artiq_browser`. They are both built-in Qt defaults, after all. See also the different scrolling behavior in the Waveform tab and experiment argument widgets depending on modifier keys. IMO the most natural place would be as part of a "getting started" step-by-step tutorial.