m-labs/artiq #638: Call method from getattr in kernel

r-srinivas, 2016-12-08:
Using 2.0 on windows, is there anyway of calling methods defined by getattr in the kernel? I tried getattr on its own in the kernel and that didn't work, and then tried using it in build instead. However, I get an error message when I try to access the method. Using the following code,

```python
from artiq.experiment import *

class Foo:

    def __init__(self, x):
        self.x = x

    def bar(self, y):
        print(self.x*y)

class Getattr_Test(EnvExperiment):

    def build(self):
        self.setattr_device("core")
        self.f = Foo(5)
        self.test_method = getattr(self.f, "bar")

    @kernel
    def run(self):
        self.test_method(10)
```

I got the message,

```
ERROR:worker(27342,getattr_test.py):root:Terminating with exception (AttributeEr
ror: type object 'Getattr_Test' has no attribute 'test_method')
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\master\worker
_impl.py", line 228, in main
    exp_inst.run()
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\language\core
.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\coredevice\co
re.py", line 111, in run
    self.compile(function, args, kwargs, set_result)
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\coredevice\co
re.py", line 95, in compile
    library = target.compile_and_link([module])
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\compiler\targ
ets.py", line 180, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules]
,
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\compiler\targ
ets.py", line 180, in <listcomp>
    return self.link([self.assemble(self.compile(module)) for module in modules]
,
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\compiler\targ
ets.py", line 137, in compile
    llmod = module.build_llvm_ir(self)
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\compiler\modu
le.py", line 82, in build_llvm_ir
    return llvm_ir_generator.process(self.artiq_ir, attribute_writeback=True)
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\compiler\tran
sforms\llvm_ir_generator.py", line 408, in process
    self.process_function(func)
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\compiler\tran
sforms\llvm_ir_generator.py", line 570, in process_function
    llinsn = getattr(self, "process_" + type(insn).__name__)(insn)
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\compiler\tran
sforms\llvm_ir_generator.py", line 1463, in process_Quote
    return self._quote(insn.value, insn.type, lambda: [repr(insn.value)])
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\compiler\tran
sforms\llvm_ir_generator.py", line 1410, in _quote
    lambda: path() + ['__class__'])
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\compiler\tran
sforms\llvm_ir_generator.py", line 1411, in _quote
    return _quote_attributes()
  File "C:\Anaconda3\envs\artiq-2016-09-27\lib\site-packages\artiq\compiler\tran
sforms\llvm_ir_generator.py", line 1388, in _quote_attributes
    attrvalue = getattr(value, attr)
AttributeError: type object 'Getattr_Test' has no attribute 'test_method'
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 27342
```

If I get rid of the kernel decorator it prints the value 50 as you expect. Is there a better way of calling methods by name in the kernel?

whitequark, 2016-12-08:
`getattr` is not supported in kernels by design. Every use of `getattr` that could be in principle efficiently compiled can also be represented using closures, which also shows intent better IMO. Is there a reason you cannot rewrite what you want using closures?

sbourdeauducq, 2016-12-08:
@whitequark He is not using ``getattr`` inside the kernel, and I don't see a technical reason for his use case not to be supported.

whitequark, 2016-12-09:
@sbourdeauducq oh, I misread that. I agree that this should work. I will look into it.

commit 2f37b1d1 (2017-01-30): compiler: support methods defined on singleton instances.

Fixes #638.
commit 2dc8d417 (2017-02-01): compiler: support methods defined on singleton instances.

Fixes #638.
closed 2017-01-30 by whitequark