m-labs/artiq #639: Determining if an edge was rise or fall from gate_both

vontell, 2016-12-08:
The `gate_both(self, duration)` method, or `_set_sensitivity(3)` configuration for a TTLInOut object returns timestamps at which a rising or falling edge has been detected. However, is there any way to determine if the edge was a rise or fall, such as a tag which gets sent with the timestamp?

sbourdeauducq, 2016-12-08:
It's alternating rises and falls, so all you really need is the initial value. What do you need that for?

vontell, 2016-12-08:
My current approach is also to use the initial value. However, how would I get this initial value; is there a TTLInOut method which would allow me to query the current input state of the channel? desired functionality is at issue [vontell/artiq-control#7](https://github.com/vontell/artiq-control/issues/7)

sbourdeauducq, 2016-12-09:
In 3.x you can use this:
https://m-labs.hk/artiq/manual-master/_modules/artiq/coredevice/ttl.html#TTLInOut.sample_input
But doing what you want without any race condition is not completely straightforward (e.g. what happens if an edge arrives at the same time as when you are sampling the first time?)